coolsnowwolf/lede #10418: 更新源的时候有两处报错 是删除还有有其他替换的

abcdefg1225, 2022-11-14:
root@OpenWrt:~# opkg update
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/x86_64/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/oui/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/oui                                                   /Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/oui/Package                                                   s.gz, wget returned 8.

上面这两处报错要怎么处理 是直接删掉这两处不需要么？

1054009064, 2022-11-14:
忽略。

abcdefg1225, 2022-11-14:
> 放电。
谢谢

closed 2022-11-14 by abcdefg1225