coolsnowwolf/lede #10424: 编译x86版本一直在编译lm-sensors模块时报错 cannot find -liconv

zsedc800, 2022-11-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有搜到带解决方案的issue

## 2. 详细叙述
系统是桌面版的Ubuntu 22.04 直接拉取的master分支代码进行编译，重试过很多次都是lm-sensors这个模块出现问题
dl 文件夹里面是存在libiconv的，不太明白链接器是搜索路径是怎样的，一直提示/home/zsedc800/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

make[3]: Entering directory '/home/zsedc800/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
x86_64-openwrt-linux-musl-gcc  -o prog/sensors/sensors prog/sensors/main.ro prog/sensors/chips.ro -liconv -Llib -lsensors
/home/zsedc800/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[3]: *** [prog/sensors/Module.mk:43: prog/sensors/sensors] Error 1
make[3]: Leaving directory '/home/zsedc800/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
make[2]: *** [Makefile:129: /home/zsedc800/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built] Error 2
make[2]: Leaving directory '/home/zsedc800/lede/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.15#0.05#0.18
    ERROR: package/feeds/packages/lm-sensors failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/lm-sensors/compile] Error 1
make[1]: Leaving directory '/home/zsedc800/lede'
make: *** [/home/zsedc800/lede/include/toplevel.mk:230：package/feeds/packages/lm-sensors/compile] 错误 2

1582130940, 2022-11-15:
https://github.com/1582130940/OpenWrt-Packages/commit/563c4e2bf63628360c8470d48633c4b671975844
你pick这个试试

zsedc800, 2022-11-16:
> [1582130940/OpenWrt-Packages@563c4e2](https://github.com/1582130940/OpenWrt-Packages/commit/563c4e2bf63628360c8470d48633c4b671975844) 你pick这个试试

还是不行，同样的报错。我在feeds/packages/utils/lm-sensors/Makefile里面加了这段重新编译还是失败
执行make dirclean和make distclean 重新操作一遍后的结果也是这个错误。

guoshh1978, 2022-11-16:
把lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/prog/sensors/Module.mk里面的
LIBICONV := $(shell if /sbin/ldconfig -p | grep -q '/libiconv\.so$$' ; then echo `\-liconv`; else echo; fi)
去掉\-liconv，改成下面的内容试试，我都是这样解决的
LIBICONV := $(shell if /sbin/ldconfig -p | grep -q '/libiconv\.so$$' ; then echo; else echo; fi)

zsedc800, 2022-11-16:
> 把lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/prog/sensors/Module.mk里面的 LIBICONV := (shellif/sbin/ldconfig−p|grep−q′/libiconv.so$' ; then echo `\-liconv`; else echo; fi) 去掉-liconv，改成下面的内容试试，我都是这样解决的 LIBICONV := (shellif/sbin/ldconfig−p|grep−q′/libiconv.so$' ; then echo; else echo; fi)

非常感谢，编译通过了。

closed 2023-04-25 by aiamadeus