coolsnowwolf/lede #10434: Package kmod-rxrpc is missing dependencies for the following libraries: ip6_udp_tunnel.ko udp_tunnel.ko

edwardostara, 2022-11-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Package kmod-rxrpc is missing dependencies for the following libraries:
ip6_udp_tunnel.ko
udp_tunnel.ko

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：Package kmod-rxrpc is missing dependencies for the following libraries:
ip6_udp_tunnel.ko
udp_tunnel.ko
modules/netsupport.mk:1155: /home/home/lede/bin/targets/x86/64/packages/kmod-rxrpc_5.15.78-1_x86_64.ipk
----------------------------------------------------------------------------------------------------------------------------------
解决方法：
编辑文件netsupport.mk，路径在./package/kernel/linux/modules/netsupport.mk
找到1155行，发现是关于rxrpc的包
将依赖改写成，如下：
define KernelPackage/rxrpc
  SUBMENU:=$(NETWORK_SUPPORT_MENU)
  TITLE:=AF_RXRPC support
  DEPENDS:= \
    +kmod-crypto-manager \
    +kmod-crypto-pcbc \
    +kmod-crypto-fcrypt \
	+kmod-udptunnel4 \
	+IPV6:kmod-udptunnel6
  HIDDEN:=1
  KCONFIG:= \
	CONFIG_AF_RXRPC \
	CONFIG_RXKAD=m \
	CONFIG_AF_RXRPC_DEBUG=n
  FILES:= \
	$(LINUX_DIR)/net/rxrpc/rxrpc.ko
  AUTOLOAD:=$(call AutoLoad,30,rxrpc.ko)
endef

define KernelPackage/rxrpc/description
  Kernel support for AF_RXRPC; required for AFS client
endef

$(eval $(call KernelPackage,rxrpc))
----------------------------------------------------------------------------------------------------------------------------------
麻烦，开发者，百忙之余把源码改一下，感谢。

coolsnowwolf, 2022-11-17:
你发个 PR 可好？

edwardostara, 2022-11-17:
> 你发个 PR 可好？

去查了下怎么pr，已提交，你看看可以不。