coolsnowwolf/lede #10439: 无法生成升级固件

anbylan, 2022-11-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法生成sysupgrade.bin文件

### (2) 路由器型号和固件版本

A：优酷路由宝L2

### (3) 详细日志

A：WARNING: Image file /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin is too big: 16777220 > 16449536
echo '{  "metadata_version": "1.1", "compat_version": "1.1", "compat_message": "Config cannot be migrated from swconfig to DSA", "new_supported_devices": ["youku,yk-l2"], "supported_devices": ["youku,yk-l2 - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA"],  "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r5414-524dfad99", "target": "ramips/mt7621", "board": "youku_yk-l2" } }' | fwtool -I - /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin
Failed to open firmware file
make[5]: [Makefile:218: /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin] Error 1 (ignored)
sha256sum "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin" | cut -d" " -f1 > "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.sha256sum"
sha256sum: /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin: No such file or directory
[ ! -s "/home/m/lede/key-build" -o ! -s "/home/m/lede/key-build.ucert" -o ! -s "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin" ] || { cp "/home/m/lede/key-build.ucert" "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.ucert" ; usign -S -m "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin" -s "/home/m/lede/key-build" -x "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.sig" ; ucert -A -c "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.ucert" -x "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.sig" ; fwtool -S "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin.ucert" "/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin" ; }
cp /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin /home/m/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin
cp: cannot stat '/home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin': No such file or directory
make[5]: [Makefile:216: /home/m/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin] Error 1 (ignored)
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/m/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips TMPDIR=/home/m/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips/tmp /home/m/lede/staging_dir/host/bin/opkg --offline-root /home/m/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch mipsel_24kc:200 list-installed > /home/m/lede/bin/targets/ramips/mt7621/openwrt-ramips-mt7621-youku_yk-l2.manifest

hxlls, 2022-11-17:
A：WARNING: Image file /home/m/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-youku_yk-l2-squashfs-sysupgrade.bin is too big: 16777220 > 16449536
 已经提示了啊，你的固件太大了

anbylan, 2022-11-17:
明白了，已经修改了，谢谢！

closed 2022-11-17 by aiamadeus

anbylan, 2022-11-19:
老大，不好意思，麻烦你，请问怎么修改才可以生成较大空间的插件，哪里修改，超出了16M