coolsnowwolf/lede #10440: conntrack-tools编译报错

quanjindeng, 2022-11-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：conntrack-tools编译报错

### (2) 路由器型号和固件版本

A：x86平台，最新源码

### (3) 详细日志

A：

make[1]: Entering directory '/home/deng/lede'
make[2]: Entering directory '/home/deng/lede/package/libs/toolchain'
echo "libc" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/deng/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.10#0.03#0.13
make[2]: Entering directory '/home/deng/lede/package/libs/libmnl'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libmnl-1.0.5.tar.bz2.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libmnl-1.0.5.tar.bz2" "274b9b919ef3152bfb3da3a13c950dd60d6e2bcd54230ffeca298d03b40d0525" "" "http://www.netfilter.org/projects/libmnl/files" "ftp://ftp.netfilter.org/pub/libmnl"    '
echo "libmnl" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libmnl.default.install
make[2]: Leaving directory '/home/deng/lede/package/libs/libmnl'
time: package/libs/libmnl/compile#0.10#0.04#0.13
make[2]: Entering directory '/home/deng/lede/feeds/packages/libs/libnetfilter-cthelper'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libnetfilter_cthelper-1.0.0.tar.bz2.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libnetfilter_cthelper-1.0.0.tar.bz2" "07618e71c4d9a6b6b3dc1986540486ee310a9838ba754926c7d14a17d8fccf3d" "" "http://www.netfilter.org/projects/libnetfilter_cthelper/files/" "ftp://ftp.netfilter.org/pub/libnetfilter_cthelper/"    '
echo "libnetfilter-cthelper" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libnetfilter-cthelper.default.install
make[2]: Leaving directory '/home/deng/lede/feeds/packages/libs/libnetfilter-cthelper'
time: package/feeds/packages/libnetfilter-cthelper/compile#0.10#0.03#0.13
make[2]: Entering directory '/home/deng/lede/feeds/packages/libs/libnetfilter-cttimeout'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libnetfilter_cttimeout-1.0.0.tar.bz2.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libnetfilter_cttimeout-1.0.0.tar.bz2" "aeab12754f557cba3ce2950a2029963d817490df7edb49880008b34d7ff8feba" "" "http://www.netfilter.org/projects/libnetfilter_cttimeout/files/" "ftp://ftp.netfilter.org/pub/libnetfilter_cttimeout/"    '
echo "libnetfilter-cttimeout" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libnetfilter-cttimeout.default.install
make[2]: Leaving directory '/home/deng/lede/feeds/packages/libs/libnetfilter-cttimeout'
time: package/feeds/packages/libnetfilter-cttimeout/compile#0.10#0.04#0.14
make[2]: Entering directory '/home/deng/lede/package/libs/libnfnetlink'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libnfnetlink-1.0.2.tar.bz2.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libnfnetlink-1.0.2.tar.bz2" "b064c7c3d426efb4786e60a8e6859b82ee2f2c5e49ffeea640cfe4fe33cbc376" "" "http://www.netfilter.org/projects/libnfnetlink/files/" "ftp://ftp.netfilter.org/pub/libnfnetlink/"    '
echo "libnfnetlink" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libnfnetlink.default.install
make[2]: Leaving directory '/home/deng/lede/package/libs/libnfnetlink'
time: package/libs/libnfnetlink/compile#0.11#0.04#0.15
make[2]: Entering directory '/home/deng/lede/feeds/packages/libs/libnetfilter-queue'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libnetfilter_queue-1.0.5.tar.bz2.flock -c '  /home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libnetfilter_queue-1.0.5.tar.bz2" "f9ff3c11305d6e03d81405957bdc11aea18e0d315c3e3f48da53a24ba251b9f5" "" "https://www.netfilter.org/projects/libnetfilter_queue/files"    '
echo "libnetfilter-queue" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libnetfilter-queue.default.install
make[2]: Leaving directory '/home/deng/lede/feeds/packages/libs/libnetfilter-queue'
time: package/feeds/packages/libnetfilter-queue/compile#0.08#0.05#0.13
make[2]: Entering directory '/home/deng/lede/feeds/packages/libs/libtirpc'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.libtirpc-1.3.2.tar.bz2.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "libtirpc-1.3.2.tar.bz2" "e24eb88b8ce7db3b7ca6eb80115dd1284abc5ec32a8deccfed2224fc2532b9fd" "" "@SF/libtirpc"    '
echo "libtirpc" >> /home/deng/lede/staging_dir/target-x86_64_musl/pkginfo/libtirpc.default.install
make[2]: Leaving directory '/home/deng/lede/feeds/packages/libs/libtirpc'
time: package/feeds/packages/libtirpc/compile#0.13#0.04#0.16
make[2]: Entering directory '/home/deng/lede/package/firmware/linux-firmware'
mkdir -p /home/deng/lede/dl
SHELL= flock /home/deng/lede/tmp/.linux-firmware-20221012.tar.xz.flock -c '  	/home/deng/lede/scripts/download.pl "/home/deng/lede/dl" "linux-firmware-20221012.tar.xz" "e9d174af729511c8cccb60ec4e0b223b3c44b67d813b42d1ab9813acfa667fa5" "" "@KERNEL/linux/kernel/firmware"    '
[... 298 lines cut ...]
conntrack.c:2537:39: note: (near initialization for 'attr2name[14]')
conntrack.c:2538:39: warning: initialized field overwritten [-Woverride-init]
 2538 |                 [CTA_STATS_MAX + 2] = "unknown2",
      |                                       ^~~~~~~~~~
conntrack.c:2538:39: note: (near initialization for 'attr2name[15]')
conntrack.c: In function 'do_command_ct':
conntrack.c:3258:51: error: 'NFCT_FILTER_DUMP_STATUS' undeclared (first use in this function); did you mean 'NFCT_FILTER_DUMP_MAX'?
 3258 |                                                   NFCT_FILTER_DUMP_STATUS,
      |                                                   ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                   NFCT_FILTER_DUMP_MAX
make[6]: *** [Makefile:696: conntrack.o] Error 1
make[6]: Leaving directory '/home/deng/lede/build_dir/target-x86_64_musl/conntrack-tools-1.4.7/src'
make[5]: *** [Makefile:738: all-recursive] Error 1
make[5]: Leaving directory '/home/deng/lede/build_dir/target-x86_64_musl/conntrack-tools-1.4.7/src'
make[4]: *** [Makefile:515: all] Error 2
make[4]: Leaving directory '/home/deng/lede/build_dir/target-x86_64_musl/conntrack-tools-1.4.7/src'
make[3]: *** [Makefile:541: all-recursive] Error 1
make[3]: Leaving directory '/home/deng/lede/build_dir/target-x86_64_musl/conntrack-tools-1.4.7'
make[2]: *** [Makefile:83: /home/deng/lede/build_dir/target-x86_64_musl/conntrack-tools-1.4.7/.built] Error 2
make[2]: Leaving directory '/home/deng/lede/feeds/packages/net/conntrack-tools'
time: package/feeds/packages/conntrack-tools/compile#0.16#0.05#0.20
    ERROR: package/feeds/packages/conntrack-tools failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/conntrack-tools/compile] Error 1
make[1]: Leaving directory '/home/deng/lede'
make: *** [/home/deng/lede/include/toplevel.mk:230：package/feeds/packages/conntrack-tools/compile] 错误 2

1582130940, 2022-11-17:
https://github.com/coolsnowwolf/lede/pull/10442
pick这两个commit试下

quanjindeng, 2022-11-17:
> #10442 pick这两个commit试下

感谢，现在编译通过了

closed 2022-11-17 by quanjindeng