coolsnowwolf/lede #10447: 网卡偶发断流, 接口重启, 内核日志显示 netifd: segfault

ricky9w, 2022-11-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

网卡偶发性断流, `状态-概览` 页面显示看不到网卡, 手动尝试重启 WAN 后问题解决

### (2) 路由器型号和固件版本

CPU: Intel N5095
NIC: Intel i225-v b3

PVE 虚拟化环境:
+ pve-manager/7.2-11/b76d3178
+ Linux 5.19.7-2-pve SMP PREEMPT_DYNAMIC PVE 5.19.7-2

Lede 版本:
+ 固件: OpenWrt R22.7.7 / LuCI Master (git-22.200.61437-0c3c82f)
+ 内核: 5.15.56

PVE 中配置了网卡直通, 光猫和 AP 均连接到直通的网口

### (3) 详细日志

系统日志太长这里省略一些重复部分, 完整版日志请参阅 gist: https://gist.github.com/ricky9w/a3c0a6bfe6adc736d0f3a46cb24c50dc

内核日志:
```
[193902.436087] netifd[1935]: segfault at 7fdc67b48098 ip 000000000041bbd6 sp 00007ffc3dc65900 error 4 in netifd[404000+1b000]
[193902.437573] Code: 01 75 31 48 8b 53 58 be fe 3b 42 00 bf 02 00 00 00 31 c0 e8 01 e6 ff ff 48 89 df e8 23 f0 ff ff c7 83 90 00 00 00 00 00 00 00 8b 7b 68 5b e9 f1 cf ff ff 5b c3 53 48 8d 9f 48 ff ff ff 83 fa
[193907.509469] br-lan: port 2(eth2) entered disabled state
[193907.510528] br-lan: port 1(eth0) entered disabled state
[193907.515712] device eth3 left promiscuous mode
[193907.516764] br-lan: port 3(eth3) entered disabled state
[193907.536369] device eth2 left promiscuous mode
[193907.537567] br-lan: port 2(eth2) entered disabled state
[193907.557992] device eth0 left promiscuous mode
[193907.559029] br-lan: port 1(eth0) entered disabled state
[193907.688088] 8021q: adding VLAN 0 to HW filter on device eth0
[193907.690101] br-lan: port 1(eth0) entered blocking state
[193907.691567] br-lan: port 1(eth0) entered disabled state
[193907.692786] device eth0 entered promiscuous mode
[193907.695377] br-lan: port 1(eth0) entered blocking state
[193907.696474] br-lan: port 1(eth0) entered forwarding state
[193907.820639] br-lan: port 2(eth2) entered blocking state
[193907.821709] br-lan: port 2(eth2) entered disabled state
[193907.822906] device eth2 entered promiscuous mode
[193907.824089] br-lan: port 2(eth2) entered blocking state
[193907.825454] br-lan: port 2(eth2) entered forwarding state
[193907.948470] br-lan: port 3(eth3) entered blocking state
[193907.949550] br-lan: port 3(eth3) entered disabled state
[193907.951454] device eth3 entered promiscuous mode
[193908.520414] br-lan: port 2(eth2) entered disabled state
[193909.500835] igc 0000:00:11.0 eth2: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
[193909.503082] br-lan: port 2(eth2) entered blocking state
[193909.504227] br-lan: port 2(eth2) entered forwarding state
[193911.488627] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[193912.028269] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[193932.319654] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[193932.859398] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[193951.183014] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[193951.722621] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[193970.010199] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[193970.549835] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[... 386 lines cut ...]
Fri Nov 18 01:40:16 2022 daemon.info pppd[28267]: CHAP authentication succeeded: Authentication Successful,Welcome!
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: CHAP authentication succeeded
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: peer from calling number D4:C1:C8:8B:DE:60 authorized
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: local IP address 10.1.69.108
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: remote IP address 10.1.64.1
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: primary DNS address 112.4.0.55
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: secondary DNS address 221.131.143.69
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Interface 'wan' is now up
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: local LL address fe80::02e2:69a8:7d55:8661
Fri Nov 18 01:40:16 2022 daemon.notice pppd[28267]: remote LL address fe80::d6c1:c8ff:fe8b:de60
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri Nov 18 01:40:16 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Fri Nov 18 01:40:16 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Nov 18 01:40:16 2022 daemon.err odhcp6c[28567]: Failed to send RS (Operation not permitted)
Fri Nov 18 01:40:20 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Nov 18 01:40:20 2022 user.notice ddns-scripts[29775]: vpn: PID '29775' started at 2022-11-18 01:40
Fri Nov 18 01:40:20 2022 daemon.notice netifd: Interface 'wan_6' is now up
Fri Nov 18 01:40:20 2022 user.info ddns-scripts[29775]: vpn: Starting main loop at 2022-11-18 01:40
Fri Nov 18 01:40:24 2022 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Fri Nov 18 01:40:26 2022 user.err ddns-scripts[29775]: vpn: cURL Error: '6'
Fri Nov 18 01:40:26 2022 user.warn ddns-scripts[29775]: vpn: Transfer failed - retry 1/0 in 60 seconds
```

coolsnowwolf, 2022-11-18:
I225 是这样的，习惯就好，硬件Bug断流

1582130940, 2022-11-18:
1、自己想办法升级NVM固件
2、自己想办法换I226
3、自己想办法换Realtek

ricky9w, 2022-11-19:
> 1、自己想办法升级NVM固件 2、自己想办法换I226 3、自己想办法换Realtek

个人而言除非换设备, 不然换网卡不太现实lol, 于是搜了一下 NVM 固件相关的问题, 在 Intel 论坛找到了这个反馈:
https://community.intel.com/t5/Ethernet-Products/Intel-Network-I225-V-Nework-issue-persist-even-after-FW-upgrade/td-p/1189493
Intel 也发了相关通告:
https://www.intel.com/content/www/us/en/support/articles/000057261/ethernet-products/gigabit-ethernet-controllers-up-to-2-5gbe.html
以及相关问题的报告:
https://cdrdv2-public.intel.com/621661/621661-Intel%C2%AE%20Ethernet%20Controller%20I225-Public%20External%20Specification%20Update-v1.2.pdf
bps
报告里提到升级固件的措施, 看了一下固件的作用似乎只是自动检测丢包问题, 在大量丢包出现时自动将速率设置为 1000Mbps, 也就是说其实没有从根本上解决问题? 报告里似乎也只是说 "mitigate" 而不是 "resolve"...

> The impact of this IPG issue can be mitigated when connected through a switch/router by setting the 
link speed in the driver (Windows or Linux) to 1Gpbs operation to enable a 1Gpbs line rate.

不过照这么说把网卡速率设置为自定义是不是也能一定程度上缓解问题?

closed 2026-09-05 by coolsnowwolf