coolsnowwolf/lede #10452: 和目AX18添加NSS-ECM-64编译失败

npc1024, 2022-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.15内核添加QCA-NSS-ECM-64编译失败

### (2） 详细日志

A：
ERROR: modpost: "ipv6_dev_find_and_hold" [/home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/qca-nss-ecm-64-2021-04-29-c115aec3/ecm.ko] undefined!
ERROR: modpost: "ndisc_send_ns" [/home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/qca-nss-ecm-64-2021-04-29-c115aec3/ecm.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:133: /home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/qca-nss-ecm-64-2021-04-29-c115aec3/Module.symvers] Error 1
make[5]: *** Deleting file '/home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/qca-nss-ecm-64-2021-04-29-c115aec3/Module.symvers'
make[4]: *** [Makefile:1815: modules] Error 2
make[4]: Leaving directory '/home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/linux-5.15.79'
make[3]: *** [Makefile:99: /home/jinyu/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/qca-nss-ecm-64-2021-04-29-c115aec3/.built] Error 2
make[3]: Leaving directory '/home/jinyu/lede/package/qca/nss/qca-nss-ecm-64'
time: package/qca/nss/qca-nss-ecm-64/compile#0.45#0.09#0.52
    ERROR: package/qca/nss/qca-nss-ecm-64 failed to build.
make[2]: *** [package/Makefile:116: package/qca/nss/qca-nss-ecm-64/compile] Error 1
make[2]: Leaving directory '/home/jinyu/lede'
make[1]: *** [package/Makefile:110: /home/jinyu/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jinyu/lede'
make: *** [/home/jinyu/lede/include/toplevel.mk:230：world] 错误 2

npc1024, 2022-11-19:
目测是IPV6的原因

Lancet-z, 2022-11-22:
顺便问下，编译成功的情况下，刷入能正常使用吗。我此前按照默认配置编译完成后的固件刷入后无法进入系统。网络连接获取不到ip。

zhs9527, 2023-01-05:
qca-nss-ecm-64编译失败是版本问题，用Boos4721提供的qca-nss-ecm可以编译通过。

cary-zhou, 2023-02-10:
[1100-export-ns.patch](https://github.com/coolsnowwolf/lede/files/10710073/1100-export-ns.patch)

coolsnowwolf, 2026-09-05:
核验当前 master：该 issue 针对的是旧的 5.15 `qca-nss-ecm-64` 包。当前树已统一为 `package/qca/qca-nss-ecm`，Qualcommmax 默认内核为 6.12（测试 6.18），并提供对应的 `target/linux/qualcommax/patches-6.12/0600-1-qca-nss-ecm-support-CORE.patch`，导出 `ipv6_dev_find_and_hold` 与 `ndisc_send_ns`；同时包内已有 Linux 6.18 兼容补丁。因此原始 modpost 缺符号问题已被当前内核/补丁栈覆盖，关闭为 completed。

closed 2026-09-05 by coolsnowwolf