coolsnowwolf/lede #10453: Mi R3G编译封闭无线驱动kmod-7603e时，出现DMA_MAPPING_ERROR

ifuturenan, 2022-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：小米路由器3G，在编译kmod-7603e+kmod-7615d闭源无线驱动时，出现DMA_MAPPING_ERROR报错，无法正常编译。

### (2) 路由器型号和固件版本

A：小米路由器3G master最新code

### (3) 详细日志

`make[3]: Entering directory '/home/user/lede/package/lean/mt/drivers/mt7603e'
rm -f /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built
touch /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built_check
make -C "/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.224" V=1 CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" M="/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap"   CONFIG_RLT_WIFI=y CONFIG_MT_WIFI_PATH="rlt_wifi" CONFIG_FIRST_IF_MT7603E=y   CONFIG_FIRST_IF_EEPROM_FLASH=y CONFIG_RT_FIRST_CARD_EEPROM="flash"  CONFIG_MULTI_INF_SUPPORT=y CONFIG_WSC_INCLUDED=y  CONFIG_DOT11N_DRAFT3=y  CONFIG_DOT11W_PMF_SUPPORT=y CONFIG_TXBF_SUPPORT=y        CONFIG_NEW_RATE_ADAPT_SUPPORT=y     CONFIG_RTMP_FLASH_SUPPORT=y    CONFIG_ATE_SUPPORT=y CONFIG_RT2860V2_AP_V24_DATA_STRUCTURE=y  CONFIG_MEMORY_OPTIMIZATION=y    CONFIG_UAPSD=y            CONFIG_RTMP_PCI_SUPPORT=y   CONFIG_RLT_AP_SUPPORT=y CONFIG_RALINK_MT7603E=y CONFIG_MT_MAC=y  CONFIG_WDS_SUPPORT=y CONFIG_MBSS_SUPPORT=y   CONFIG_APCLI_SUPPORT=y CONFIG_MULTI_APCLI_SUPPORT=y CONFIG_AIRPLAY_SUPPORT=y CONFIG_MAC_REPEATER_SUPPORT=y CONFIG_DFS_SUPPORT=y  CONFIG_DOT11K_RRM_SUPPORT=y  CONFIG_LLTD_SUPPORT=y   CONFIG_MULTI_CORE_SUPPORT=y     CONFIG_MEMORY_OPTIMIZATION=y      modules
make[4]: Entering directory '/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.224'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap \
single-build= \
need-builtin=1 need-modorder=1
  mipsel-openwrt-linux-musl-gcc -Wp,-MD,/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/ap/.ap.o.d  -nostdinc -isystem /home/user/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I./arch/mips/include -I./arch/mips/include/generated  -I./include -I./arch/mips/include/uapi -I./arch/mips/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80001000 -DLINKER_LOAD_ADDRESS=0x80001000 -DDATAOFFSET=0 -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -fno-stack-check -march=mips32r2 -mtune=34kc -Wa,--trap -DTOOLCHAIN_SUPPORTS_VIRT -DTOOLCHAIN_SUPPORTS_XPA -DTOOLCHAIN_SUPPORTS_CRC -DTOOLCHAIN_SUPPORTS_DSP -DTOOLCHAIN_SUPPORTS_GINV -I./arch/mips/include/asm/mach-ralink -I./arch/mips/include/asm/mach-ralink/mt7621 -I./arch/mips/include/asm/mach-generic -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -fmacro-prefix-map=./= -Wno-packed-not-aligned -I/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include -I/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/ate/include -DMT7603 -DMT_BBP -DMT_RF -DRTMP_RF_RW_SUPPORT -DMT_MAC -DRTMP_MAC_PCI -DRTMP_PCI_SUPPORT -DMCS_LUT_SUPPORT -DRELEASE_EXCLUDE -DMT_RF -DRTMP_RF_RW_SUPPORT -DRTMP_PCI_SUPPORT -DRX_DMA_SCATTER -DCONFIG_ANDES_SUPPORT -DRTMP_EFUSE_SUPPORT -DA_BAND_SUPPORT -DNEW_MBSSID_MODE -DENHANCE_NEW_MBSSID_MODE -DRESOURCE_PRE_ALLOC -DRTMP_EFUSE_SUPPORT -DCONFIG_FPGA_MODE -DDATA_QUEUE_RESERVE -DMULTI_CLIENT_SUPPORT -DSMART_CARRIER_SENSE_SUPPORT -DLIMIT_GLOBAL_SW_QUEUE -DNEW_RATE_ADAPT_SUPPORT -DRTMP_MAC_PCI -DRTMP_PCI_SUPPORT -DUAPSD_SUPPORT -DUAPSD_DEBUG -DMT_PS -DWSC_INCLUDED -DWSC_SINGLE_TRIGGER -DWSC_AP_SUPPORT -DDOT11W_PMF_SUPPORT -DSOFT_ENCRYPT -DAPCLI_DOT11W_PMF_SUPPORT -DMTFWD -DDOT11N_DRAFT3 -DLLTD_SUPPORT -DDOT11K_RRM_SUPPORT -DAP_SCAN_SUPPORT -DSCAN_SUPPORT -DAPPLE_11K_IOT -DMBSS_SUPPORT -DWDS_SUPPORT -DAPCLI_SUPPORT -DMAT_SUPPORT -DTRAFFIC_BASED_TXOP -DAPCLI_AUTO_CONNECT_SUPPORT -DMULTI_APCLI_SUPPORT -DAIRPLAY_SUPPORT -DMAC_REPEATER_SUPPORT -DCONFIG_ATE -DCONFIG_QA -DCONFIG_RT2880_ATE_CMD_NEW -DRTMP_FLASH_SUPPORT -DMCS_LUT_SUPPORT -DTXBF_SUPPORT -DVHT_TXBF_SUPPORT -DDFS_SUPPORT -DDFS_ATP_SUPPORT -DMULTI_INF_SUPPORT -DRTMP_EFUSE_SUPPORT -DCAL_FREE_IC_SUPPORT -DCONFIG_RT_FIRST_CARD=y -DCONFIG_RT_FIRST_IF_RF_OFFSET=0 -DCONFIG_RT_FIRST_CARD_EEPROM="\"flash\"" -DCONFIG_FIRST_IF_EPAELNA -DENHANCED_STAT_DISPLAY -DNEW_RATE_ADAPT_SUPPORT -DRELEASE_EXCLUDE -DFIFO_EXT_SUPPORT -DCONFIG_BA_REORDER_MONITOR -DUSE_BMC -DMT_PS -DDMA_RESET_SUPPORT -DTXRXCR_DEBUG_SUPPORT -DMT7603_E1 -DMT7603_E2 -DEDCCA_RB -DLOAD_FW_ONE_TIME -DEAPOL_QUEUE_SUPPORT -DAGGREGATION_SUPPORT -DPIGGYBACK_SUPPORT -DWMM_SUPPORT -DLINUX -Wall -Wstrict-prototypes -Wno-trigraphs -Wframe-larger-than=4096 -DCONFIG_RALINK_FLASH_API -DCONFIG_AP_SUPPORT -DSCAN_SUPPORT -DAP_SCAN_SUPPORT -DDOT11_N_SUPPORT -DSTATS_COUNT_SUPPORT -DIAPP_SUPPORT -DDOT1X_SUPPORT -DMEMORY_OPTIMIZATION -DIP_ASSEMBLY -DACS_CTCC_SUPPORT -DSW_ATF_SUPPORT  -DMODULE -mno-long-calls  -DKBUILD_BASENAME='"ap"' -DKBUILD_MODNAME='"mt7603e"' -c -o /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/ap/ap.o /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/ap/ap.c
In file included from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rtmp_os.h:46,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rtmp_comm.h:82,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rt_config.h:35,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/ap/ap.c:29:
/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/os/rt_linux.h:886: warning: "DMA_MAPPING_ERROR" redefined
 #define DMA_MAPPING_ERROR(_handle, _ptr) \
 
In file included from ./include/linux/pci-dma-compat.h:8,
                 from ./include/linux/pci.h:2389,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/os/rt_linux.h:28,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rtmp_os.h:46,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rtmp_comm.h:82,
                 from /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/../mt7603_wifi/include/rt_config.h:35,
[... cut ...]
make[3]: Leaving directory '/home/user/lede/package/lean/mt/drivers/mt7603e'
time: package/lean/mt/drivers/mt7603e/compile#2.16#1.02#2.94
    ERROR: package/lean/mt/drivers/mt7603e failed to build.
make[2]: *** [package/Makefile:116: package/lean/mt/drivers/mt7603e/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:230: world] Error 2
`

coolsnowwolf, 2022-11-19:
并无这个问题

ifuturenan, 2022-11-19:
经过复测，发现是.config文件过老导致。

ifuturenan, 2022-11-19:
尝试使用默认.config进行编译，mt7603e配置如下：
[image]
[image]
问题依旧存在。

ifuturenan, 2022-11-20:
经过反复编译验证，MT7603E在编译时不支持 Tx Beamforming，如果选中会编译失败。取消勾选即可正常编译。

[image]

closed 2022-11-20 by ifuturenan