coolsnowwolf/lede #10457: 好像每次更dnsmasq，adbyby都有问题哦

rongcenter, 2022-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
好像每次更新dnsmasq，adbyby都有问题哦，直接断网。
### (2) 路由器型号和固件版本

x86

### (3) 详细日志

Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:27 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:27 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:28 2022 daemon.info procd: - init complete -
Sat Nov 19 22:52:32 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:32 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:32 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:32 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:37 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:37 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:37 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:37 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:42 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:42 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:42 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:42 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:47 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:47 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:47 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:47 2022 user.err : jail: failed to load dependencies

lovehackintosh, 2022-11-19:
是更新了规则后断网的吗，dnsmasq无法启动，把adbyby 关了就没事了

coolsnowwolf, 2022-11-19:
/var/etc/dnsmasq-adbyby.d 这个目录不存在

closed 2022-11-20 by rongcenter

rongcenter, 2022-11-21:
更新失败导致