coolsnowwolf/lede #10461: adbyby开启后会导致dnsmasq启用失败

firkerword, 2022-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：adbyby开启后会导致dnsmasq v2.87启用失败，不保留配置刷机和虚拟机新建机器均出现此问题，尝试手动创建目录依然无法解决

### (2) 路由器型号和固件版本

A：
机器型号：X86-3867U
固件版本：5.15.79  6.0.9固件  
package版本：R22.11.11 / LuCI Master (git-22.323.58461-c7fb305)

### (3) 详细日志

A：Sun Nov 20 21:35:52 2022 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Nov 20 21:36:04 2022 daemon.err uhttpd[2147]: cp: can't create '/tmp/etc/dnsmasq-adbyby.d/adblock': No such file or directory
Sun Nov 20 21:36:04 2022 cron.err crond[7630]: crond (busybox 1.35.0) started, log level 5
Sun Nov 20 21:36:10 2022 cron.err crond[7935]: crond (busybox 1.35.0) started, log level 5
Sun Nov 20 21:36:10 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:10 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:10 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:10 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:15 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:15 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:20 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:20 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:26 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:26 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:31 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:31 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Nov 20 21:36:36 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Nov 20 21:36:36 2022 daemon.crit dnsmasq[1]: FAILED to start up

Ginua8292, 2022-11-20:
不知道是不是这个问题！能拨号！就是打不开网站

yuzhe1129, 2022-11-20:
我也是，昨天编译的K2P固件与今天X86编译的固件也是开了就断网，关了adbyby马上有网了，什么情况，搞得我怀疑人生了，

Ginua8292, 2022-11-20:
还有分流也有问题！节点旧的固件没问题！最新源码编译的固件就是连不上！所有插件都一样

ecrasy, 2022-11-21:
建议去这些package的作者那里提issue
让他们来适配新版本的dnsmasq

starxu1971, 2022-11-22:
只要进行有关操作就无法上网，必须手动重启防火墙才行，有时候完全不能上网。

xiangfeigao, 2022-11-22:
等待修复。

samba自动共享怎么老是编译失败

firkerword, 2022-11-23:
奇怪了 dnsmasq 降级到v2.86 凌晨编译的固件还是这个问题😅

starxu1971, 2022-11-24:
目前我是用10月8号编译的固件在用，没有问题，但是dnsmasq用2.87的这个固件也不能上网了。

firkerword, 2022-11-24:
> 目前我是用10月8号编译的固件在用，dnsmasq2.86没有问题，但是升级到dnsmasq2.87后固件也不能上网了。

我之前双十一的固件也没问题   找不出是哪的问题

firkerword, 2022-11-24:
[image]
这个目录是存在的  但是读取不到

fseteen, 2022-11-25:
R5S 7.19的原版固件，有这个问题，打开 Adbyby 无法上网。但是前两天重新编译的 Adbyby 的 IPK 包解决了这个问题，很好用！

starxu1971, 2022-11-25:
昨天编译的只要打开Adbyby Plus+ 就无法上网。

nb12nb34, 2022-11-29:
哪个大佬知道怎么解决吗，目前一直困扰啊。。

ZGNCZMY, 2022-12-11:
我也遇到了，Adbyby Plus+一关就正常

sheip9, 2022-12-13:
应该不是dnsmasq的问题，参考https://github.com/immortalwrt/luci/commit/8423fc6924d4617b253e2af6dc8744078ff86b2b

closed 2022-12-19 by firkerword

wjcroom, 2023-04-24:
> 应该不是dnsmasq的问题，参考[immortalwrt/luci@8423fc6](https://github.com/immortalwrt/luci/commit/8423fc6924d4617b253e2af6dc8744078ff86b2b)

这个也许可行，但是最新的文件应该是变化了，参照这个方式修改就行了
``add_dns()
{
	mkdir -p /tmp/dnsmasq.d/dnsmasq-adbyby.d
	mkdir -p /tmp/dnsmasq.d

	[ -s $PROG_PATH/adesc.conf ] && awk '!/^$/&&!/^#/{printf("ipset=/%s/'"adbyby_esc"'\n",$0)}' $PROG_PATH/adesc.conf > /tmp/dnsmasq.d/dnsmasq-adbyby.d/06-dnsmasq.esc

	local tt="/tmp/adby_plus_d1.txt"
	local var=1
	local dcc=0
	local ipsmd5=""
	local addnsmd5=""
	if [ $wan_mode -eq 1 ]; then
		for i in ` grep -oE "[0-9]+(\.[0-9]+){3}" $PROG_PATH/adhost.conf`;do ipset -! add adbyby_wan $i;done
		grep -oE "[0-9a-zA-Z][0-9a-zA-Z\-]*(\.[0-9a-zA-Z\-]+)*\.[a-zA-Z]{2,}" $PROG_PATH/adhost.conf \
		|awk '!/^$/&&!/^#/{printf("ipset=/%s/'"adbyby_wan"'\n",$0)}' > $tt
		if [ $plus_help -eq 1 ]; then
			rm -f /tmp/plusad1.txt
			if [ -s /tmp/adbyby/data/lazy.txt ]; then
				get_domain "/tmp/adbyby/data/lazy.txt" "/tmp/plusad1.txt"
			fi
			if [ -s /tmp/adbyby/data/video.txt ]; then
				get_domain "/tmp/adbyby/data/video.txt" "/tmp/plusad1.txt"
			fi
			if [ -s $DATA_PATH/user.txt ]; then
				get_domain "$DATA_PATH/user.txt" "/tmp/plusad1.txt"
			fi
			if [ -s /tmp/plusad1.txt ];then
				awk '!/^$/&&!/^#/{printf("ipset=/%s/'"adbyby_wan"'\n",$0)}' /tmp/plusad1.txt >> $tt
			fi
		fi
		[ -s $tt ] && echo "`sort -u ${tt} 2>/dev/null`" > $WAN_FILE
		ipsmd5=`md5sum $WAN_FILE 2>/dev/null| cut -d ' ' -f1`
		if ls /tmp/dnsmasq.d/dnsmasq-adbyby.d/* >/dev/null 2>&1; then
			mkdir -p /tmp/dnsmasq.d
		fi
	fi
	dcc=`ls /tmp/dnsmasq.d/dnsmasq-adbyby.d/* 2>/dev/null | wc -l`
	if [ $dcc -gt 0 ];then
		echo 'conf-dir=/tmp/dnsmasq.d/dnsmasq-adbyby.d' > /tmp/dnsmasq.d/dnsmasq-adbyby.conf
		addnsmd5=`md5sum /tmp/dnsmasq.d/dnsmasq-adbyby.conf 2>/dev/null| cut -d ' ' -f1`
	fi
	mkdir -p /tmp/adbyby/rules/data /tmp/adbyby/rules/host
	echo "$addnsmd5|$ipsmd5" > /tmp/adbyby/addnsmd5.txt
	rm -f /tmp/plusad1.txt
	rm -f $tt
}

del_dns()
{
	rm -f /tmp/dnsmasq.d/dnsmasq-adbyby.conf
	rm -f /tmp/dnsmasq.d/dnsmasq-adbyby.d/*
	rm -f /tmp/adbyby_host.conf
}

包括开头那个地址var / etc/dnsmasq.d/