coolsnowwolf/lede #10462: open-vm-tools : 编译错误

sKyissKy, 2022-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：master branch编译 open-vm-tools失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

make[3]: Entering directory '/home/ubuntu/lede/feeds/packages/utils/open-vm-tools'
rm -f /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/.configured_*
rm -f /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.open-vm-tools_installed
(cd /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /home/ubuntu/lede/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/ubuntu/lede/staging_dir/host/bin/autom4te AUTOCONF=/home/ubuntu/lede/staging_dir/host/bin/autoconf AUTOMAKE=/home/ubuntu/lede/staging_dir/host/bin/automake ACLOCAL=/home/ubuntu/lede/staging_dir/host/bin/aclocal AUTOHEADER=/home/ubuntu/lede/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/ubuntu/lede/staging_dir/host/bin/libtoolize LIBTOOL=/home/ubuntu/lede/staging_dir/host/bin/libtool M4=/home/ubuntu/lede/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/home/ubuntu/lede/staging_dir/host/bin/libtoolize --install' /home/ubuntu/lede/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/ubuntu/lede/staging_dir/host/share/aclocal -I /home/ubuntu/lede/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/ubuntu/lede/staging_dir/hostpkg/share/aclocal -I /home/ubuntu/lede/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/ubuntu/lede/staging_dir/host/bin/aclocal -I /home/ubuntu/lede/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/ubuntu/lede/staging_dir/hostpkg/share/aclocal -I /home/ubuntu/lede/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: /home/ubuntu/lede/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `config'.
OpenWrt-libtoolize: linking file `config/config.guess'
OpenWrt-libtoolize: linking file `config/config.sub'
OpenWrt-libtoolize: linking file `config/install-sh'
OpenWrt-libtoolize: linking file `config/ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
autoreconf: running: /home/ubuntu/lede/staging_dir/host/bin/autoconf --include=/home/ubuntu/lede/staging_dir/target-x86_64_musl/host/share/aclocal --include=/home/ubuntu/lede/staging_dir/hostpkg/share/aclocal --include=/home/ubuntu/lede/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/ubuntu/lede/staging_dir/host/share/aclocal --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: /home/ubuntu/lede/staging_dir/host/bin/automake --add-missing --force-missing
lib/guestRpc/Makefile.am:26: warning: 'CFLAGS' is a user variable, you should not override it;
lib/guestRpc/Makefile.am:26: use 'AM_CFLAGS' instead
libappmonitor/Makefile.am:39: warning: source file '$(top_srcdir)/lib/stubs/stub-panic.c' is in a subdirectory,
libappmonitor/Makefile.am:39: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
libappmonitor/Makefile.am:39: warning: source file '$(libappmonitor_rpcchanneldir)/bdoorChannel.c' is in a subdirectory,
libappmonitor/Makefile.am:39: but option 'subdir-objects' is disabled
libappmonitor/Makefile.am:39: warning: source file '$(libappmonitor_rpcchanneldir)/rpcChannel.c' is in a subdirectory,
libappmonitor/Makefile.am:39: but option 'subdir-objects' is disabled
libappmonitor/Makefile.am:39: warning: source file '$(libappmonitor_rpcchanneldir)/glib_stubs.c' is in a subdirectory,
libappmonitor/Makefile.am:39: but option 'subdir-objects' is disabled
libappmonitor/Makefile.am:47: warning: source file '$(libappmonitor_rpcchanneldir)/vsockChannel.c' is in a subdirectory,
libappmonitor/Makefile.am:47: but option 'subdir-objects' is disabled
libappmonitor/Makefile.am:47: warning: source file '$(libappmonitor_rpcchanneldir)/simpleSocket.c' is in a subdirectory,
libappmonitor/Makefile.am:47: but option 'subdir-objects' is disabled
libguestlib/Makefile.am:40: warning: source file '$(libguestlib_rpcchanneldir)/bdoorChannel.c' is in a subdirectory,
libguestlib/Makefile.am:40: but option 'subdir-objects' is disabled
libguestlib/Makefile.am:40: warning: source file '$(libguestlib_rpcchanneldir)/rpcChannel.c' is in a subdirectory,
[... 76 lines cut ...]
vmblock-fuse/Makefile.am:40: but option 'subdir-objects' is disabled
vmblock-fuse/Makefile.am:40: warning: source file '$(top_srcdir)/modules/shared/vmblock/stubs.c' is in a subdirectory,
vmblock-fuse/Makefile.am:40: but option 'subdir-objects' is disabled
vmhgfs-fuse/Makefile.am:38: warning: source file '$(top_srcdir)/lib/stubs/stub-debug.c' is in a subdirectory,
vmhgfs-fuse/Makefile.am:38: but option 'subdir-objects' is disabled
vmhgfs-fuse/Makefile.am:38: warning: source file '$(top_srcdir)/lib/stubs/stub-log.c' is in a subdirectory,
vmhgfs-fuse/Makefile.am:38: but option 'subdir-objects' is disabled
vmhgfs-fuse/Makefile.am:38: warning: source file '$(top_srcdir)/lib/stubs/stub-panic.c' is in a subdirectory,
vmhgfs-fuse/Makefile.am:38: but option 'subdir-objects' is disabled
autoreconf: Leaving directory `.'
(cd /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/./; if [ -x ./configure ]; then find /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r chmod u+w; find /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r -n1 cp --remove-destination /home/ubuntu/lede/scripts/config.guess; find /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r chmod u+w; find /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r -n1 cp --remove-destination /home/ubuntu/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protec [... cut ...]
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:150: /home/ubuntu/lede/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/utils/open-vm-tools'
time: package/feeds/packages/open-vm-tools/compile#8.24#0.47#9.68
    ERROR: package/feeds/packages/open-vm-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/open-vm-tools/compile] Error 1

sKyissKy, 2022-11-21:
各位大佬看看能不能修复open-vm-tools-11.3.0的编译，如果不能修复，最好是和openwrt 官方包用12.0.0，这样可以直接编译通过。不知道当时Beginner-Go大佬为啥做了revert。

xiaoyouke01, 2022-11-21:
我也碰到了这个错误

coldrook, 2022-11-21:
我也是

sKyissKy, 2022-11-21:
> 我也碰到了这个错误

用op官方的packages的open-vm-tools 12.0.0 替换就能直接编译通过。

sKyissKy, 2022-11-21:
> 我也是

用op官方的packages的open-vm-tools 12.0.0 替换就能直接编译通过。

xiaotan8, 2022-11-21:
我也碰到了这个错误

bleach1991, 2022-11-22:
同样遇到了

xiaotan8, 2022-11-22:
大雕修复了

closed 2022-11-22 by sKyissKy