coolsnowwolf/lede #10467: softethervpn5断电重启配置丢失

qiji102, 2022-11-21:
## 1.softethervpn5断电重启配置丢失

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2.自编译lede大神的openwrt固件编译添加softethervpn5，编译成功，windows系统s使用softether server tool工具配置完成后断电重启设备配置丢失，重新编译几次也仍然是这个情况。只有使用softether server tool配置完成，然后登录路由器后台SoftEther VPN 服务器设置中点击保存应用才能保存设置，如果不登录路由器后台SoftEther VPN 服务器保存，断电重启设备，当前设置将不会被保存。

### (2) 
固件：amdx86-64，vm虚拟机运行
固件版本 ：OpenWrt R22.11.11 / LuCl Master git-22.316.26068-65f106a)
内核版本：5.15.78
softethervpn版本：SoftEtherVPN-5.02.5180

qiji102, 2022-11-21:
日志文件：
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.231363] pci 0000:00:17.3: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.232826] pci 0000:00:16.7: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.234192] pci 0000:00:16.7: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.235692] pci 0000:00:16.6: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.237032] pci 0000:00:16.6: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.238600] pci 0000:00:16.5: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.240020] pci 0000:00:16.5: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.241395] pci 0000:00:16.4: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.242763] pci 0000:00:16.4: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.244326] pci 0000:00:16.3: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.246018] pci 0000:00:16.3: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.248089] pci 0000:00:15.7: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.249711] pci 0000:00:15.7: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.251201] pci 0000:00:15.6: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.252941] pci 0000:00:15.6: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.254572] pci 0000:00:15.5: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.255941] pci 0000:00:15.5: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.257344] pci 0000:00:15.4: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.258812] pci 0000:00:15.4: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.260209] pci 0000:00:15.3: BAR 7: no space for [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.261543] pci 0000:00:15.3: BAR 7: failed to assign [io  size 0x1000]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.263318] pci 0000:00:01.0: PCI bridge to [bus 01]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.264575] pci 0000:02:01.0: BAR 6: assigned [mem 0xfd500000-0xfd50ffff pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.266169] pci 0000:00:11.0: PCI bridge to [bus 02]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.267340] pci 0000:00:11.0:   bridge window [io  0x2000-0x3fff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.268909] pci 0000:00:11.0:   bridge window [mem 0xfd500000-0xfdffffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.270806] pci 0000:00:11.0:   bridge window [mem 0xe7b00000-0xe7ffffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.272473] pci 0000:00:15.0: PCI bridge to [bus 03]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.273618] pci 0000:00:15.0:   bridge window [io  0x4000-0x4fff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.274981] pci 0000:00:15.0:   bridge window [mem 0xfd400000-0xfd4fffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.276437] pci 0000:00:15.0:   bridge window [mem 0xe7a00000-0xe7afffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.278270] pci 0000:00:15.1: PCI bridge to [bus 04]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.279514] pci 0000:00:15.1:   bridge window [io  0x8000-0x8fff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.280956] pci 0000:00:15.1:   bridge window [mem 0xfd000000-0xfd0fffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.282551] pci 0000:00:15.1:   bridge window [mem 0xe7600000-0xe76fffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.284385] pci 0000:00:15.2: PCI bridge to [bus 05]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.285736] pci 0000:00:15.2:   bridge window [io  0xc000-0xcfff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.287430] pci 0000:00:15.2:   bridge window [mem 0xfcc00000-0xfccfffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.288859] pci 0000:00:15.2:   bridge window [mem 0xe7200000-0xe72fffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.290588] pci 0000:00:15.3: PCI bridge to [bus 06]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.291937] pci 0000:00:15.3:   bridge window [mem 0xfc800000-0xfc8fffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.293670] pci 0000:00:15.3:   bridge window [mem 0xe6e00000-0xe6efffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.295479] pci 0000:00:15.4: PCI bridge to [bus 07]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.296774] pci 0000:00:15.4:   bridge window [mem 0xfc400000-0xfc4fffff]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.298403] pci 0000:00:15.4:   bridge window [mem 0xe6a00000-0xe6afffff 64bit pref]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.300130] pci 0000:00:15.5: PCI bridge to [bus 08]
Tue Nov 22 01:54:54 2022 kern.info kernel: [    1.301447] pci 0000:00:15.5:   bridge window [mem 0xfc000000-0xfc0fffff]
[... cut ...]
Tue Nov 22 01:55:02 2022 daemon.notice ttyd[3443]: [2022/11/22 01:55:02:5824] N:  Listening on port: 7681
Tue Nov 22 01:55:07 2022 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Tue Nov 22 01:55:07 2022 daemon.info procd: - init complete -
Tue Nov 22 01:55:10 2022 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Nov 22 01:55:10 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Tue Nov 22 01:55:26 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)

hellcatjack, 2022-12-05:
安装softethervpn是选择这个吗？luci/application/luci-app-softethervpn 
理论上该组件每5分钟自动保存配置的，可以在日志中看到类似这种提示：
Mon Dec  5 13:36:46 2022 daemon.notice softethervpn-server[12875]: Writing vpn_server.config (AutoSaveConfigSpan set to < 300 > seconds)

closed 2026-09-05 by coolsnowwolf