coolsnowwolf/lede #10468: 大佬帮忙看看为什么编译失败

hxxh2017, 2022-11-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：[409/567] Processing param_functions.h: script/generate_param.py bin/default/docs-xml/smbdotconf/parameters.all.xml -> bin/default/lib/param/param_functions.h
[410/567] Processing param_local.h: script/generate_param.py bin/default/docs-xml/smbdotconf/parameters.all.xml -> bin/default/lib/param/param_local.h
[411/567] Processing param_global.h: script/generate_param.py bin/default/docs-xml/smbdotconf/parameters.all.xml -> bin/default/lib/param/param_global.h
[412/567] Processing param_table_gen.c: script/generate_param.py bin/default/docs-xml/smbdotconf/parameters.all.xml -> bin/default/lib/param/param_table_gen.c
[413/567] Processing librpc/idl/drsuapi.idl
[414/567] Processing librpc/idl/epmapper.idl
librpc/idl/epmapper.idl:203: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/epmapper.idl:204: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/epmapper.idl:219: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead

[415/567] Processing librpc/idl/eventlog.idl
[416/567] Processing librpc/idl/browser.idl
[417/567] Processing librpc/idl/dssetup.idl
librpc/idl/dssetup.idl:84: warning: top-level [out] pointer `info' is not a [ref] pointer

[418/567] Processing librpc/idl/frsapi.idl
librpc/idl/drsuapi.idl:1907: warning: helper() is pidl-specific and deprecated. Use `include' instead
librpc/idl/drsuapi.idl:757: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/drsuapi.idl:761: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/drsuapi.idl:769: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/drsuapi.idl:777: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/drsuapi.idl:785: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/drsuapi.idl:793: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead

[419/567] Processing librpc/idl/spoolss.idl
[420/567] Processing librpc/idl/dnsserver.idl
librpc/idl/dnsserver.idl:1530: warning: helper() is pidl-specific and deprecated. Use `include' instead
librpc/idl/dnsserver.idl:1462: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/dnsserver.idl:1518: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead

[421/567] Processing librpc/idl/samr.idl
[422/567] Processing librpc/idl/srvsvc.idl
librpc/idl/spoolss.idl:3567: warning: helper() is pidl-specific and deprecated. Use `include' instead
librpc/idl/spoolss.idl:919: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:1126: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:1318: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:1669: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:1693: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:1768: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:2116: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:2507: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/spoolss.idl:892: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1126: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1139: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1318: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1641: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1669: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1693: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1731: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:1768: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2116: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2135: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2231: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2273: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2469: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:2507: warning: top-level [out] pointer `info' is not a [ref] pointer
librpc/idl/spoolss.idl:3014: warning: [out] argument `info' not a pointer
librpc/idl/spoolss.idl:3107: warning: top-level [out] pointer `info' is not a [ref] pointer

[423/567] Processing librpc/idl/winreg.idl
librpc/idl/srvsvc.idl:1459: warning: top-level [out] pointer `hnd' is not a [ref] pointer

[424/567] Processing librpc/idl/mgmt.idl
librpc/idl/winreg.idl:191: warning: Got pointer for `size', expected fully dereferenced variable
librpc/idl/winreg.idl:191: warning: Got pointer for `length', expected fully dereferenced variable
librpc/idl/winreg.idl:191: warning: Got pointer for `size', expected fully dereferenced variable
librpc/idl/winreg.idl:191: warning: Got pointer for `length', expected fully dereferenced variable
librpc/idl/winreg.idl:191: warning: Got pointer for `length', expected fully dereferenced variable
librpc/idl/winreg.idl:191: warning: Got pointer for `length', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_size', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_length', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_size', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_length', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_length', expected fully dereferenced variable
librpc/idl/winreg.idl:268: warning: Got pointer for `data_length', expected fully dereferenced variable

[425/567] Processing librpc/idl/netlogon.idl
[426/567] Processing librpc/idl/svcctl.idl
librpc/idl/svcctl.idl:674: warning: helper() is pidl-specific and deprecated. Use `include' instead
librpc/idl/svcctl.idl:456: warning: top-level [out] pointer `TagId' is not a [ref] pointer
librpc/idl/svcctl.idl:467: warning: top-level [out] pointer `service_status' is not a [ref] pointer

[427/567] Processing librpc/idl/wkssvc.idl
[428/567] Processing librpc/idl/eventlog6.idl
librpc/idl/netlogon.idl:1803: warning: helper() is pidl-specific and deprecated. Use `include' instead
librpc/idl/netlogon.idl:830: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/netlogon.idl:848: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead
librpc/idl/netlogon.idl:1092: warning: subcontext() is deprecated. Use represent_as() or transmit_as() instead

[429/567] Processing librpc/idl/backupkey.idl
librpc/idl/backupkey.idl:154: warning: helper() is pidl-specific and deprecated. Use `include' instead

[430/567] Processing librpc/idl/witness.idl
[431/567] Processing librpc/idl/clusapi.idl
Segmentation fault

Waf: Leaving directory `/home/hxxh/lede/build_dir/target-x86_64_musl/samba-4.14.14/bin/default'
Build failed
 -> task in 'PIDL_WITNESS' failed with exit status 139 (run with -v to display more information)
make[3]: *** [Makefile:454: /home/hxxh/lede/build_dir/target-x86_64_musl/samba-4.14.14/.built] Error 1
make[3]: Leaving directory '/home/hxxh/lede/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/compile#102.62#22.45#103.94
    ERROR: package/feeds/packages/samba4 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/samba4/compile] Error 1
make[2]: Leaving directory '/home/hxxh/lede'
make[1]: *** [package/Makefile:110: /home/hxxh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hxxh/lede'
make: *** [/home/hxxh/lede/include/toplevel.mk:231：world] 错误 2

1582130940, 2022-11-21:
Segmentation fault

jeasonliang, 2022-11-22:
make menuconfig---network---
[image]

closed 2026-09-05 by coolsnowwolf