coolsnowwolf/lede #10474: 最近编译x86版一直遇到gnutls出错

iKira, 2022-11-22:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [√] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近一周编译x86版都会遇到gnutls错误，编译rockchip就可以正常通过。麻烦协助排查一下是gnutls哪方面的原因导致编译失败，thx。

### (2) 路由器型号和固件版本

A：x86，AMD R7 5825U，用EXSi开虚拟机

### (3) 详细日志

A：

.libs/libcmd-danetool.lax/libcerttool-cfg.a/sys_socket.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/threadlib.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/time_rz.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/timegm.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/timespec.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/unistd.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/vasnprintf.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/xalloc-die.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/xmalloc.o .libs/libcmd-danetool.lax/libcerttool-cfg.a/xsize.o .libs/libcmd-danetool.lax/lt2-asnprintf.o .libs/libcmd-danetool.lax/lt3-basename-lgpl.o .libs/libcmd-danetool.lax/lt4-bitrotate.o .libs/libcmd-danetool.lax/lt5-c-ctype.o .libs/libcmd-danetool.lax/lt6-c-strcasecmp.o .libs/libcmd-danetool.lax/lt7-c-strncasecmp.o .libs/libcmd-danetool.lax/lt8-cloexec.o .libs/libcmd-danetool.lax/lt9-close-stream.o .libs/libcmd-danetool.lax/lt10-error.o .libs/libcmd-danetool.lax/lt11-exitfail.o .libs/libcmd-danetool.lax/lt12-fcntl.o .libs/libcmd-danetool.lax/lt13-fd-hook.o .libs/libcmd-danetool.lax/lt14-free.o .libs/libcmd-danetool.lax/lt15-getdelim.o .libs/libcmd-danetool.lax/lt16-getline.o .libs/libcmd-danetool.lax/lt17-getprogname.o .libs/libcmd-danetool.lax/lt18-gettime.o .libs/libcmd-danetool.lax/lt19-gl_linked_list.o .libs/libcmd-danetool.lax/lt20-gl_list.o .libs/libcmd-danetool.lax/lt21-gl_xlist.o .libs/libcmd-danetool.lax/lt22-hash-pjw-bare.o .libs/libcmd-danetool.lax/lt23-hash.o .libs/libcmd-danetool.lax/lt24-lock.o .libs/libcmd-danetool.lax/lt25-malloca.o .libs/libcmd-danetool.lax/lt26-memchr.o .libs/libcmd-danetool.lax/lt27-mktime.o .libs/libcmd-danetool.lax/lt28-nstrftime.o .libs/libcmd-danetool.lax/lt29-parse-datetime.o .libs/libcmd-danetool.lax/lt30-printf-args.o .libs/libcmd-danetool.lax/lt31-printf-parse.o .libs/libcmd-danetool.lax/lt32-progname.o .libs/libcmd-danetool.lax/lt33-read-file.o .libs/libcmd-danetool.lax/lt34-sockets.o .libs/libcmd-danetool.lax/lt35-stat-time.o .libs/libcmd-danetool.lax/lt36-sys_socket.o .libs/libcmd-danetool.lax/lt37-threadlib.o .libs/libcmd-danetool.lax/lt38-time_rz.o .libs/libcmd-danetool.lax/lt39-timegm.o .libs/libcmd-danetool.lax/lt40-timespec.o .libs/libcmd-danetool.lax/lt41-unistd.o .libs/libcmd-danetool.lax/lt42-vasnprintf.o .libs/libcmd-danetool.lax/lt43-xalloc-die.o .libs/libcmd-danetool.lax/lt44-xmalloc.o .libs/libcmd-danetool.lax/lt45-xsize.o .libs/libcmd-danetool.lax/lt46-asnprintf.o .libs/libcmd-danetool.lax/lt47-basename-lgpl.o .libs/libcmd-danetool.lax/lt48-bitrotate.o .libs/libcmd-danetool.lax/lt49-c-ctype.o .libs/libcmd-danetool.lax/lt50-c-strcasecmp.o .libs/libcmd-danetool.lax/lt51-c-strncasecmp.o .libs/libcmd-danetool.lax/lt52-cloexec.o .libs/libcmd-danetool.lax/lt53-dirname-lgpl.o .libs/libcmd-danetool.lax/lt54-fcntl.o .libs/libcmd-danetool.lax/lt55-fd-hook.o .libs/libcmd-danetool.lax/lt56-free.o .libs/libcmd-danetool.lax/lt57-getdelim.o .libs/libcmd-danetool.lax/lt58-getline.o .libs/libcmd-danetool.lax/lt59-hash-pjw-bare.o .libs/libcmd-danetool.lax/lt60-hash.o .libs/libcmd-danetool.lax/lt61-lock.o .libs/libcmd-danetool.lax/lt62-malloca.o .libs/libcmd-danetool.lax/lt63-memchr.o .libs/libcmd-danetool.lax/lt64-printf-args.o .libs/libcmd-danetool.lax/lt65-printf-parse.o .libs/libcmd-danetool.lax/lt66-read-file.o .libs/libcmd-danetool.lax/lt67-stat-time.o .libs/libcmd-danetool.lax/lt68-stripslash.o .libs/libcmd-danetool.lax/lt69-sys_socket.o .libs/libcmd-danetool.lax/lt70-threadlib.o .libs/libcmd-danetool.lax/lt71-unistd.o .libs/libcmd-danetool.lax/lt72-vasnprintf.o .libs/libcmd-danetool.lax/lt73-xsize.o .libs/libcmd-danetool.lax/libminitasn1.a/coding.o .libs/libcmd-danetool.lax/libminitasn1.a/decoding.o .libs/libcmd-danetool.lax/libminitasn1.a/element.o .libs/libcmd-danetool.lax/libminitasn1.a/errors.o .libs/libcmd-danetool.lax/libminitasn1.a/gstr.o .libs/libcmd-danetool.lax/libminitasn1.a/parser_aux.o .libs/libcmd-danetool.lax/libminitasn1.a/structure.o .libs/libcmd-danetool.lax/libminitasn1.a/version.o .libs/libcmd-danetool.lax/lt74-asnprintf.o .libs/libcmd-danetool.lax/lt75-basename-lgpl.o .libs/libcmd-danetool.lax/lt76-bitrotate.o .libs/libcmd-danetool.lax/lt77-c-ctype.o .libs/libcmd-danetool.lax/lt78-c-strcasecmp.o .libs/libcmd-danetool.lax/lt79-c-strncasecmp.o .libs/libcmd-danetool.lax/lt80-cloexec.o .libs/libcmd-danetool.lax/lt81-dirname-lgpl.o .libs/libcmd-danetool.lax/lt82-fcntl.o .libs/libcmd-danetool.lax/lt83-fd-hook.o .libs/libcmd-danetool.lax/lt84-free.o .libs/libcmd-danetool.lax/lt85-getdelim.o .libs/libcmd-danetool.lax/lt86-getline.o .libs/libcmd-danetool.lax/lt87-hash-pjw-bare.o .libs/libcmd-danetool.lax/lt88-hash.o .libs/libcmd-danetool.lax/lt89-lock.o .libs/libcmd-danetool.lax/lt90-malloca.o .libs/libcmd-danetool.lax/lt91-memchr.o .libs/libcmd-danetool.lax/lt92-printf-args.o .libs/libcmd-danetool.lax/lt93-printf-parse.o .libs/libcmd-danetool.lax/lt94-read-file.o .libs/libcmd-danetool.lax/lt95-stat-time.o .libs/libcmd-danetool.lax/lt96-stripslash.o .libs/libcmd-danetool.lax/lt97-sys_socket.o .libs/libcmd-danetool.lax/lt98-threadlib.o .libs/libcmd-danetool.lax/lt99-unistd.o .libs/libcmd-danetool.lax/lt100-vasnprintf.o .libs/libcmd-danetool.lax/lt101-xsize.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/malloca.o .libs/libcmd-certtool.lax/lt90-malloca.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/malloca.o .libs/libcmd-certtool.lax/lt90-malloca.o
/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/memchr.o .libs/libcmd-certtool.lax/lt91-memchr.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/memchr.o .libs/libcmd-certtool.lax/lt91-memchr.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/printf-args.o .libs/libcmd-certtool.lax/lt92-printf-args.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/printf-args.o .libs/libcmd-certtool.lax/lt92-printf-args.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/printf-parse.o .libs/libcmd-certtool.lax/lt93-printf-parse.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/printf-parse.o .libs/libcmd-certtool.lax/lt93-printf-parse.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/read-file.o .libs/libcmd-certtool.lax/lt94-read-file.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/read-file.o .libs/libcmd-certtool.lax/lt94-read-file.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/stat-time.o .libs/libcmd-certtool.lax/lt95-stat-time.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/stat-time.o .libs/libcmd-certtool.lax/lt95-stat-time.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/stripslash.o .libs/libcmd-certtool.lax/lt96-stripslash.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/stripslash.o .libs/libcmd-certtool.lax/lt96-stripslash.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/sys_socket.o .libs/libcmd-certtool.lax/lt97-sys_socket.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/sys_socket.o .libs/libcmd-certtool.lax/lt97-sys_socket.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/threadlib.o .libs/libcmd-certtool.lax/lt98-threadlib.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/threadlib.o .libs/libcmd-certtool.lax/lt98-threadlib.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/unistd.o .libs/libcmd-certtool.lax/lt99-unistd.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/unistd.o .libs/libcmd-certtool.lax/lt99-unistd.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/vasnprintf.o .libs/libcmd-certtool.lax/lt100-vasnprintf.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/vasnprintf.o .libs/libcmd-certtool.lax/lt100-vasnprintf.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt1-libgnu.a/xsize.o .libs/libcmd-certtool.lax/lt101-xsize.o || cp .libs/libcmd-certtool.lax/lt1-libgnu.a/xsize.o .libs/libcmd-certtool.lax/lt101-xsize.o
OpenWrt-libtool: link: ln .libs/libcmd-certtool.lax/lt2-libgnu_gpl.a/asnprintf.o .libs/libcmd-certtool.lax/lt102-asnprintf.o || cp .libs/libcmd-certtool.lax/lt2-libgnu_gpl.a/asnprintf.o .libs/libcmd-certtool.lax/lt102-asnprintf.o
[... cut ...]
echo "libgnutls" >> /home/ubuntu/lede/staging_dir/target-x86_64_musl/pkginfo/gnutls.default.install
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#273.83#27.46#155.71
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

iKira, 2022-11-24:
最新源码已经解决这个问题。

closed 2022-11-24 by iKira

xyz201704, 2024-08-21:
我刚好和你是反的