coolsnowwolf/lede #10476: 最新的两版源码编译后国内网页有些打不开，打开也慢

yishan2019, 2022-11-22:
我用的HK1旁路由，最新的是22号下载编译的源码，次新的是大约17，18号下载的。无论使用和不使用SSPPLUS+，都是如此，国内网站打开不流畅，有的一些网页还打不开（比如点击网页里的汽车类和其他类的链接图标）

ecrasy, 2022-11-22:
好像是dns请求的问题
你试试重启dnsmasq看看
/etc/init.d/dnsmasq restart

yishan2019, 2022-11-22:
> 我过一会刷上这个版本实验一下，结果告诉你

yishan2019, 2022-11-22:
>
>
>
>> 好像是dns请求的问题
>> 你试试重启dnsmasq看看
>> /etc/init.d/dnsmasq restart
>>
>> 我过一会刷上这个版本实验一下，结果告诉你
>>
> 重启了dns
> masa，仍然不行，win10自带的浏览器主页下面页面里的东方头条介绍车的，还有网页里易有料这两个链接里介绍车的页面都打不开，还有工商银行的也比较困难，
其他的有的勉强有的打不开，总的来说不翻墙上网的那些网站都不好用

yishan2019, 2022-11-22:
> 重启了dnsmasq.
> 还是不行，win10自带的浏览器主页下面页面里的东方头条介绍车的，还有网页里易有料这两个链接里介绍车的页面都打不开，还有工商银行的也比较困难，

yishan2019, 2022-11-22:
重启了dnsmasq，仍然不行，win10自带的浏览器主页下面页面里的东方头条介绍车的，还有网页里易有料这两个链接里介绍车的页面都打不开，还有工商银行的也比较困难，www.qq.com 腾讯主页也不行，打开后字变得很大还不全，其他的有的勉强有的打不开，总的来说不翻墙上网的那些网站都不好用

yishan2019, 2022-11-22:
OpenWrt R22.11.11 / LuCI Master (git-22.316.26068-65f106a) 这个版本和今天新下载补丁的版本都是不好用的
这个版本是目前最好的：OpenWrt R22.11.11 / LuCI Master (git-22.289.58375-2a380a3)。现在易有料打不开是他网站的问题，但多数时间我测试时打不开是旁路由固件的问题。应为我把旁路由取消掉了直接用主路由打开都很通畅。

yishan2019, 2022-11-22:
编译选项时似乎去掉了Docker后好一点，正在重复测试，明天告诉你结果

Ginua8292, 2022-11-22:
别说了最新的编出来的固件！只要涉及到防火墙全部网站都打不开了，passwall SSR 都试了！全部崩掉！最后连国内网站都打不开！

yishan2019, 2022-11-22:
经过两次编译的固件对比，编译时如果选了Docker选项，则国内有许多网站都打不开或者不流畅，不选还好一些。但是翻墙似乎不受影响

starxu1971, 2022-11-23:
只要进行有关操作就无法上网，必须手动重启防火墙才行，有时候完全不能上网。

starxu1971, 2022-11-23:
> > 别说了最新的编出来的固件！只要涉及到防火墙全部网站都打不开了，passwall SSR 都试了！全部崩掉！最后连国内网站都打不开！
> 
> 不至于吧 x86 昨天编译的 除了需要重启 dnsmasq 其它都挺正常
但是你只要开启了自动更新节点什么就又会出问题了

xiaoyouke01, 2022-11-23:
确实好多网站都打不开了

snyzf, 2022-11-24:
确实是好多网站打不开。dns返回很多网址Ip就是127.0.0.1.或者是没有ip。。重新编译一下看看。

ufk119, 2022-11-24:
已回退到2.86版本了，今天中午编译的固件看没有问题。前两天的确实有问题，我还以为是我乱加了些什么东西导致冲突呢。

starxu1971, 2022-11-27:
最新的源码退回2.86了，但SSR应用设置后还是要手动重启防火墙才行。

jerichon, 2022-11-27:
> 最新的源码退回2.86了，但SSR应用设置后还是要手动重启防火墙才行。
好像开启了byby plus还是一样上不了网

nb12nb34, 2022-11-27:
昨天编译，还是不能上网，能获取到IP地址

closed 2023-02-28 by aiamadeus