coolsnowwolf/lede #10482: 编译失败

FanxJK, 2022-11-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

```
../../../../build_dir/hostpkg/glib-2.74.0/meson.build:1925: WARNING: DEPRECATION: Option 'force_posix_threads' is deprecated and will be removed after GLib 2.72; please file an issue with your use case if you still require it
Run-time dependency threads found: YES
Header <pthread.h> has symbol "pthread_attr_setstacksize" : YES
Header <pthread.h> has symbol "pthread_attr_setinheritsched" : YES
Header <pthread.h> has symbol "pthread_condattr_setclock" : YES
Header <pthread.h> has symbol "pthread_cond_timedwait_relative_np" : NO
Header <pthread.h> has symbol "pthread_getname_np" : YES
Header <sys/syscall.h> has symbol "SYS_sched_getattr" : YES
Checking if "pthread_setname_np(const char*)" with dependency threads: links: NO
Checking if "pthread_setname_np(pthread_t, const char*)" with dependency threads: links: YES
Checking if "stack grows check" runs: NO (1)
Run-time dependency iconv found: YES
Looking for a fallback subproject for the dependency libpcre2-8 because:
Use of fallback dependencies is forced.
Subproject exists but has no meson.build file
Subproject  pcre2 is buildable: NO (disabling)
Dependency libpcre2-8 from subproject pcre2 found: NO (subproject failed to configure)
Looking for a fallback subproject for the dependency libpcre2-8 because:
Use of fallback dependencies is forced.

../../../../build_dir/hostpkg/glib-2.74.0/meson.build:2062:2: ERROR: Subproject "subprojects/pcre2" required but not found.

A full log can be found at /home/ubuntu/lede/build_dir/hostpkg/glib-2.74.0/openwrt-build/meson-logs/meson-log.txt
make[3]: *** [Makefile:120: /home/ubuntu/lede/build_dir/hostpkg/glib-2.74.0/.configured] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#4.95#1.24#6.25
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/glib2/host/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2
```

lovehackintosh, 2022-11-23:
缺少依赖pcre2

FanxJK, 2022-11-23:
> 缺少依赖pcre2

检查了pcre2，发现已安装。wsl环境
$ pcre2-config --version
10.39

FanxJK, 2022-11-23:
WSL 系统由 Ubuntu 22.04 更换为 Debian 11，已正常

closed 2022-11-23 by FanxJK

hzjnet, 2022-11-29:
回退上一个版本的glib2就正常了