coolsnowwolf/lede #10483: ipv6.无法正常分配使用

oceans-wu, 2022-11-23:
请教个问题，困扰了好几天，我用的 版本是 5.10.100 其中更换过几个固件结果都是一样，我的架构是在路由器下面接软路由（openwrt），wan口HDCP获取地址，lan口在DHCP分配，下面PC。上游下来直接接PC能正常自动获取ipv4和ipv6,在软路由（openwrt）wan口能正常获取到ipv4和ipv6，但在lan口却无法获取ipv6，软路由下游无法获取ipv6地址。DHCP6  选择服务器模式的是可以获取配置的ipv6 DNS 地址没有ipv6地址。三个模式都试过了，还是不行。请教是哪里出的问题。补充DHCP高级禁止解析没有勾选，负载均衡策略备用成员为默认

lqzhj, 2022-12-01:
同问

xkelvinx666, 2022-12-02:
编译时勾选extra/ipv6-helper即可，默认lan里就是配置好的，内置ipv6管理都不用动。
只需要手动去dhcp/dns里把过滤禁止解析 IPv6 DNS 记录去掉就行。
然后测试发现lan口的DHCP 服务器租期不要动，不然会导致部分设备ipv6获取失败。在我这表现从12h变成10m导致iphone+pc可以吗，但是ps5获取失败，而且dhcp6列表不显示。改回12h即正常

nining, 2023-05-29:
> 编译时勾选extra/ipv6-helper即可，默认lan里就是配置好的，内置ipv6管理都不用动。
This worked.

coolsnowwolf, 2026-09-05:
该 issue 没有提供目标型号、`/etc/config/network`/`dhcp`、`ubus call network.interface.wan6 status` 或 odhcp6c 日志，无法证明是 LEDE 代码缺陷。评论已经给出并验证可行的配置方向：编译启用 `ipv6helper`，LAN 保持 DHCPv6/RA 服务器模式，取消“禁止解析 IPv6 DNS 记录”，不要把 LAN DHCP 租期改成过短值（评论中改成 10m 会导致部分设备无法获取 IPv6，恢复 12h 后正常）。这是配置/运营商前缀委派问题，当前无可合并代码修复，关闭 issue。

closed 2026-09-05 by coolsnowwolf