coolsnowwolf/lede #10485: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory

xiangfeigao, 2022-11-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [1 ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：广告屏蔽大师 Plus+启用后，客户端无法上网

### (2) 路由器型号和固件版本

A：mt7621

### (3) 详细日志

A：Wed Nov 23 15:08:54 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 3c:42:7e:01:65:d5
Wed Nov 23 15:09:19 2022 cron.err crond[6581]: crond (busybox 1.35.0) started, log level 5
Wed Nov 23 15:09:26 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:26 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:31 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:31 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:36 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:36 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:41 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:41 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:47 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:47 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:52 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:09:52 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:09:52 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Wed Nov 23 15:14:26 2022 cron.err crond[7260]: crond (busybox 1.35.0) started, log level 5
Wed Nov 23 15:14:33 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:33 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:38 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:38 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:44 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:44 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:49 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:49 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:54 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:54 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:59 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Wed Nov 23 15:14:59 2022 daemon.crit dnsmasq[1]: FAILED to start up
Wed Nov 23 15:14:59 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Wed Nov 23 15:15:13 2022 cron.err crond[7794]: crond (busybox 1.35.0) started, log level 5

xiangfeigao, 2022-11-23:
/文件目录是存在的，为什么还有这个提示/难道是权限问题

coolsnowwolf, 2022-11-23:
/var/etc/dnsmasq-adbyby.d 这个文件夹

rongcenter, 2022-11-26:
规则更新失败导致的。更新成功就没有问题，我试过

closed 2022-11-26 by coolsnowwolf