coolsnowwolf/lede #10486: 编译mbedtls出错

ufk119, 2022-11-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 有类似issue，
  发生在r2s上 commit 80a7e79a7b2e5a9245d4f650f9f6a0f2c9c22517，不过之前的问题是makefile中脚本有笔误。我看我的是最新版本，已经修复了的。

## 2. 详细叙述

### (1) 具体问题

A：编译mbedtls出错，详见log

### (2) 路由器型号和固件版本

A：nanopi-r5s

### (3) 详细日志

A：
make[3]: Entering directory '/media/mark/Umjb/Openwrt/lede_R5S/lede/feeds/packages/libs/c-ares'
mkdir -p /media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /media/mark/Umjb/Openwrt/lede_R5S/lede/tmp/.root-copy.flock -c 'cp -fpR /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/c-ares-1.18.1/.pkgdir/libcares/. /media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libcares_installed
echo "libcares" >> /media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/target-aarch64_generic_musl/pkginfo/c-ares.default.install
make[3]: Leaving directory '/media/mark/Umjb/Openwrt/lede_R5S/lede/feeds/packages/libs/c-ares'
time: package/feeds/packages/c-ares/compile#0.17#0.17#0.32
make[3]: Entering directory '/media/mark/Umjb/Openwrt/lede_R5S/lede/package/libs/mbedtls'
rm -f /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/.built
touch /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/.built_check
MAKEFLAGS="" /media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/host/bin/ninja  -j1 -C /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1 
ninja: Entering directory `/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1'
[1/299] Building C object library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o
_**FAILED: library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o**_ 
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc  -I/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/include -I/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1=mbedtls-2.28.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -O2 -fPIC -MD -MT library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -MF library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o.d -o library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -c /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c: In function 'mbedtls_armv8ce_aes_crypt_ecb':
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10913:1: error: inlining failed in call to 'always_inline' 'vaesmcq_u8': target specific option mismatch
10913 | vaesmcq_u8 (uint8x16_t data)
      | ^~~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:62:17: note: called from here
   62 |             x = vaesmcq_u8( x );
      |                 ^~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10899:1: error: inlining failed in call to 'always_inline' 'vaeseq_u8': target specific option mismatch
10899 | vaeseq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:61:17: note: called from here
   61 |             x = vaeseq_u8( x, k );
      |                 ^~~~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10913:1: error: inlining failed in call to 'always_inline' 'vaesmcq_u8': target specific option mismatch
10913 | vaesmcq_u8 (uint8x16_t data)
      | ^~~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:62:17: note: called from here
   62 |             x = vaesmcq_u8( x );
      |                 ^~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10899:1: error: inlining failed in call to 'always_inline' 'vaeseq_u8': target specific option mismatch
10899 | vaeseq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:61:17: note: called from here
   61 |             x = vaeseq_u8( x, k );
      |                 ^~~~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10899:1: error: inlining failed in call to 'always_inline' 'vaeseq_u8': target specific option mismatch
10899 | vaeseq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:66:13: note: called from here
[... 32 lines cut ...]
10906 | vaesdq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:74:17: note: called from here
   74 |             x = vaesdq_u8( x, k );
      |                 ^~~~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10906:1: error: inlining failed in call to 'always_inline' 'vaesdq_u8': target specific option mismatch
10906 | vaesdq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:79:13: note: called from here
   79 |         x = vaesdq_u8( x, k );
      |             ^~~~~~~~~~~~~~~~~
In file included from /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:30:
/media/mark/Umjb/Openwrt/lede_R5S/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include/arm_neon.h:10906:1: error: inlining failed in call to 'always_inline' 'vaesdq_u8': target specific option mismatch
10906 | vaesdq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
/media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:79:13: note: called from here
   79 |         x = vaesdq_u8( x, k );
      |             ^~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:176: /media/mark/Umjb/Openwrt/lede_R5S/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/.built] Error 1
make[3]: Leaving directory '/media/mark/Umjb/Openwrt/lede_R5S/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.45#0.23#0.67
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1

ufk119, 2022-11-23:
当然临时解决办法就是取消勾选Enable use of the ARMv8 Crypto Extensions，可以正常编译。

ecrasy, 2022-11-24:
可以不取消
修改makefile
把对mbedtls的依赖改为openssl也可以

ufk119, 2022-12-07:
就此关闭，不再纠结了。

closed 2022-12-07 by ufk119