coolsnowwolf/lede #10489: 疑似广告屏蔽大师 Plus+与i226网卡冲突

Darks41, 2022-11-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
J4125+i226软路由使用广告屏蔽大师 Plus+导致路由器无法分配DHCP，也无法上网。

### (1) 具体问题

A：在CWWK家买的J4125+i226软路由，刷好自编译的Lean大固件后发现拨号正常，能显示出公网IP，从内网任一客户端都可以登录路由后台，软路由后台显示一条DHCP租约都没有，AP后台显示无法获取上一级路由的网络，内网中的客户端也无法上网，后经排查疑似是广告屏蔽大师 Plus+引起的故障，关掉插件后一切正常，开启插件后故障又会出现。

### (2) 路由器型号和固件版本

A：CWWK J4125+i226，OpenWrt R22.11.11 / LuCI Master (git-22.326.45811-f7b0b5e)

### (3) 详细日志

A：
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.552558] Non-volatile memory driver v1.3
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.556811] Linux agpgart interface v0.103
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.562136] loop: module loaded
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.565314] Guest personality initialized and is inactive
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.570744] VMCI host device registered (name=vmci, major=10, minor=127)
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.577457] Initialized host personality
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.581499] VMware PVSCSI driver - version 1.0.7.0-k
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.586495] hv_vmbus: registering driver hv_storvsc
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.591805] nvme nvme0: pci function 0000:01:00.0
Thu Nov 24 14:47:19 2022 kern.debug kernel: [    2.596591] ahci 0000:00:12.0: version 3.0
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.597012] ahci 0000:00:12.0: AHCI 0001.0301 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.602656] nvme nvme0: missing or invalid SUBNQN field.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.605128] ahci 0000:00:12.0: flags: 64bit ncq sntf pm clo only pmp pio slum part deso sadm sds apst
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.611590] nvme nvme0: allocated 64 MiB host memory buffer.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.620059] scsi host0: ahci
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.628711] scsi host1: ahci
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.631672] ata1: SATA max UDMA/133 abar m2048@0xa1619000 port 0xa1619100 irq 124
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.634899] nvme nvme0: 4/0/0 default/read/poll queues
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.639172] ata2: SATA max UDMA/133 abar m2048@0xa1619000 port 0xa1619180 irq 124
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.650844] GPT:Primary header thinks Alt. header is not at the end of the disk.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.651982] VMware vmxnet3 virtual NIC driver - version 1.6.0.0-k-NAPI
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.659231] GPT:852512 != 234441647
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.665785] hv_vmbus: registering driver hv_netvsc
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.669272] GPT:Alternate GPT header not at the end of the disk.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.674072] Fusion MPT base driver 3.04.20
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.680094] GPT:852512 != 234441647
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.684207] Copyright (c) 1999-2008 LSI Corporation
Thu Nov 24 14:47:19 2022 kern.warn kernel: [    2.687714] GPT: Use GNU Parted to correct GPT errors.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.692611] Fusion MPT SPI Host driver 3.04.20
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.697771]  nvme0n1: p1 p2 p128
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.702230] Fusion MPT SAS Host driver 3.04.20
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.709932] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.716473] ehci-pci: EHCI PCI platform driver
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.720940] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.727140] ohci-pci: OHCI PCI platform driver
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.731611] uhci_hcd: USB Universal Host Controller Interface driver
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.738201] xhci_hcd 0000:00:15.0: xHCI Host Controller
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.743448] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 1
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.751938] xhci_hcd 0000:00:15.0: hcc params 0x200077c1 hci version 0x100 quirks 0x0000000000009810
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.761310] xhci_hcd 0000:00:15.0: xHCI Host Controller
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.766551] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 2
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.773959] xhci_hcd 0000:00:15.0: Host supports USB 3.0 SuperSpeed
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.780601] hub 1-0:1.0: USB hub found
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.784410] hub 1-0:1.0: 9 ports detected
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.790472] hub 2-0:1.0: USB hub found
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.794299] hub 2-0:1.0: 7 ports detected
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.799643] usbcore: registered new interface driver usb-storage
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.805714] i8042: PNP: No PS/2 controller found.
Thu Nov 24 14:47:19 2022 kern.info kernel: [    2.810428] hv_vmbus: registering driver hyperv_keyboard
[... 523 lines cut ...]
Thu Nov 24 14:47:26 2022 daemon.notice pppd[3409]: remote IP address 121.33.112.1
Thu Nov 24 14:47:26 2022 daemon.notice pppd[3409]: primary   DNS address 202.96.128.166
Thu Nov 24 14:47:26 2022 daemon.notice pppd[3409]: secondary DNS address 202.96.134.133
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 121.33.114.143.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: New relevant interface pppoe-wan.IPv4 for mDNS.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Registering new address record for 121.33.114.143 on pppoe-wan.IPv4.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Withdrawing address record for 121.33.114.143 on pppoe-wan.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 121.33.114.143.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 121.33.114.143.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: New relevant interface pppoe-wan.IPv4 for mDNS.
Thu Nov 24 14:47:26 2022 daemon.info avahi-daemon[2250]: Registering new address record for 121.33.114.143 on pppoe-wan.IPv4.
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Interface 'wan' is now up
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Thu Nov 24 14:47:26 2022 daemon.notice netifd: Interface 'wan_6' is now up
Thu Nov 24 14:47:26 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Thu Nov 24 14:47:27 2022 cron.err crond[5051]: crond (busybox 1.35.0) started, log level 8
Thu Nov 24 14:47:27 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Thu Nov 24 14:47:27 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Nov 24 14:47:27 2022 daemon.err nmbd[4473]: [2022/11/24 14:47:27.431279,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Thu Nov 24 14:47:27 2022 daemon.err nmbd[4473]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections

nb12nb34, 2022-11-25:
同样问题，畅网4125-i225网卡一样不能获取IP地址，wan口拨号正常

Darks41, 2022-11-25:
更新：换回ARM架构的R5S（RK3568+RTL8125B）后开启Adbyby一切正常，从其他人的issue来看似乎是dnsmasq的锅，跟i226网卡没太大关系

closed 2023-02-28 by aiamadeus