coolsnowwolf/lede #10492: ERROR: package/feeds/packages/irqbalance failed to build.

zhangguanzhang, 2022-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r4s 编译失败

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
make[2]: Entering directory '/workdir/openwrt/feeds/packages/utils/irqbalance'
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/.built_check
CFLAGS="-O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result   -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0=irqbalance-1.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include   " CXXFLAGS="-O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result   -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0=irqbalance-1.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include   " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib   -znow -zrelro " make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result   -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0=irqbalance-1.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
make  all-recursive
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
Making all in tests
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/tests'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/tests'
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
/bin/bash ./libtool  --tag=CC   --mode=link ccache_cc  -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/glib-2.0 -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib/glib-2.0/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/include  -O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result   -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0=irqbalance-1.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib   -znow -zrelro  -o irqbalance activate.o bitmap.o classify.o cputree.o irqbalance.o irqlist.o numa.o placement.o procinterrupts.o  /workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib/libglib-2.0.a  -liconv -lm 
OpenWrt-libtool: link: ccache_cc -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/glib-2.0 -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib/glib-2.0/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/include -O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0=irqbalance-1.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -znow -zrelro -o irqbalance activate.o bitmap.o classify.o cputree.o irqbalance.o irqlist.o numa.o placement.o procinterrupts.o  -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib /workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib/libglib-2.0.a -liconv -lm
/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:561: irqbalance] Error 1
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
make[4]: *** [Makefile:648: all-recursive] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
make[3]: *** [Makefile:457: all] Error 2
make[3]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0'
make[2]: *** [Makefile:83: /workdir/openwrt/build_dir/target-aarch64_generic_musl/irqbalance-1.8.0/.built] Error 2
make[2]: Leaving directory '/workdir/openwrt/feeds/packages/utils/irqbalance'
time: package/feeds/packages/irqbalance/compile#0.20#0.06#0.28
    ERROR: package/feeds/packages/irqbalance failed to build.
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:116: package/feeds/packages/irqbalance/compile] Error 1
make: *** [/workdir/openwrt/include/toplevel.mk:231: package/compile] Error 2
```

KevinMX, 2022-11-26:
/home/mx/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv

我这边也复现了，工具链找不到 iconv。目前似乎只有 x86_64 的工具链受影响，ramips/mt7621, ipq40xx 暂时没事。

zhangguanzhang, 2022-11-27:
> 

有解决吗

KevinMX, 2022-11-27:
https://github.com/coolsnowwolf/lede/issues/10424

KevinMX, 2022-11-27:
https://github.com/openwrt/packages/commit/59cadc7c65d5540b3e1bba26dfcd782cad98a5fe

上游主线似乎已经移除了对 libiconv 的依赖，也许这边的 irqbalance 包也应该跟进更新？

（或者这个 issue 应该开在 packages 那边吗）

zhangguanzhang, 2022-11-30:
我发现骷髅头的irqbalance和lede的package版本是一样的，骷髅头的不修改能编译通过，感觉和gcc也可能有关系

KevinMX, 2022-12-04:
找到问题原因了。
问题出在这个 PR。Revert 之后就好了。
https://github.com/coolsnowwolf/lede/pull/10454

GreatMichaelLee, 2022-12-06:
这个问题没有在这里的仓库主干修复啊？

KevinMX, 2022-12-08:
> 这个问题没有在这里的仓库主干修复啊？

没有。

@lovehackintosh 麻烦再检查下 https://github.com/coolsnowwolf/lede/pull/10454 ，直到现在依旧无法通过编译。

GreatMichaelLee, 2022-12-08:
> > 这个问题没有在这里的仓库主干修复啊？
> 
> 没有。
> 
> @lovehackintosh 麻烦再检查下 #10454 ，直到现在依旧无法通过编译。

一样， #10565

lovehackintosh, 2022-12-08:
libiconv已经被libiconv-full取代
把libiconv替换成libiconv-full试试

zhangguanzhang, 2022-12-13:
貌似没问题了

closed 2022-12-13 by zhangguanzhang