coolsnowwolf/lede #10493: 6.0编译错误。

sjkhsl, 2022-11-25:
6.0內核编译错误。

```
In file included from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/compiler.h:3,
                 from ./include/asm-generic/bug.h:5,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/asm-generic/bug.h:3,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/bug.h:3,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/linux/average.h:5,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/compat-hacks.h:109,
                 from <command-line>:
./include/linux/compiler.h:248: note: this is the location of the previous definition
  248 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_elp.o
In file included from ./include/linux/slab.h:16,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/slab.h:3,
                 from ./include/linux/fs.h:45,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/fs.h:3,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/skbuff.h:17,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/if_ether.h:3,
                 from ./include/linux/etherdevice.h:20,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/etherdevice.h:3,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/main.h:209,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_elp.h:10,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_elp.c:7:
/home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/linux/overflow.h:32: warning: "is_signed_type" redefined
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      |
In file included from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/compiler.h:3,
                 from ./include/asm-generic/bug.h:5,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/asm-generic/bug.h:3,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/bug.h:3,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/linux/average.h:5,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/compat-hacks.h:109,
                 from <command-line>:
./include/linux/compiler.h:248: note: this is the location of the previous definition
  248 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_ogm.o
In file included from ./include/linux/slab.h:16,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/slab.h:3,
                 from ./include/linux/fs.h:45,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/fs.h:3,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/skbuff.h:17,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/if_ether.h:3,
                 from ./include/linux/etherdevice.h:20,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/etherdevice.h:3,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/main.h:209,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_ogm.h:10,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bat_v_ogm.c:7:
/home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/linux/overflow.h:32: warning: "is_signed_type" redefined
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      |
In file included from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/compiler.h:3,
                 from ./include/asm-generic/bug.h:5,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/asm-generic/bug.h:3,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
[... 62 lines cut ...]
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/linux/bug.h:3,
                 from /home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/linux/average.h:5,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/compat-hacks.h:109,
                 from <command-line>:
./include/linux/compiler.h:248: note: this is the location of the previous definition
  248 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      |
/home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bridge_loop_avoidance.c: In function 'batadv_bla_send_claim':
/home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bridge_loop_avoidance.c:446:9: error: implicit declaration of function 'netif_rx_any_context' [-Werror=implicit-function-declaration]
  446 |         netif_rx_any_context(skb);
      |         ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:249: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv/bridge_loop_avoidance.o] Error 1
make[4]: *** [Makefile:1852: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/net/batman-adv] Error 2
make[4]: Leaving directory '/home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.0.9'
make[3]: *** [Makefile:95: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.0/.built] Error 2
make[3]: Leaving directory '/home/sjkhsl/lede/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#3.39#0.36#3.74
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/lede'
make: *** [/home/sjkhsl/lede/include/toplevel.mk:231: world] Error 2
```

1054009064, 2022-11-26:
请更新 feeds/routing 进行测试

iKira, 2022-11-27:
你自己先试试默认不勾选任何插件的前提下编译是否通过，能通过的话说明你勾选的插件并不兼容6.0内核，自己按照日志挨个排查到底是哪个插件有问题。

closed 2022-12-27 by aiamadeus