coolsnowwolf/lede #10496: ERROR: please fix package 是什么错误？

yuzhe1129, 2022-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

刚git pull  以为编译固件的，make menuconfig就出现几行错误命令了，请问这个怎么解决

### (3) 详细日志

nstalling all packages from feed telephony.
Installing all packages from feed oui.
xxoo@ubuntu:~/lede$ rm -rf ./tmp && rm -rf .config
xxoo@ubuntu:~/lede$ make menuconfig
ERROR: please fix package/feeds/oui/oui-app-acl/Makefile - see logs/package/feeds/oui/oui-app-acl/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-backup/Makefile - see logs/package/feeds/oui/oui-app-backup/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-demo/Makefile - see logs/package/feeds/oui/oui-app-demo/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-dhcp-lease/Makefile - see logs/package/feeds/oui/oui-app-dhcp-lease/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-home/Makefile - see logs/package/feeds/oui/oui-app-home/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-layout/Makefile - see logs/package/feeds/oui/oui-app-layout/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-login/Makefile - see logs/package/feeds/oui/oui-app-login/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-stations/Makefile - see logs/package/feeds/oui/oui-app-stations/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-system/Makefile - see logs/package/feeds/oui/oui-app-system/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-upgrade/Makefile - see logs/package/feeds/oui/oui-app-upgrade/dump.txt for details
ERROR: please fix package/feeds/oui/oui-app-user/Makefile - see logs/package/feeds/oui/oui-app-user/dump.txt for details
ERROR: please fix package/feeds/oui/oui-ui-core/Makefile - see logs/package/feeds/oui/oui-ui-core/dump.txt for details
Collecting package info: done
Collecting target info: done

ecrasy, 2022-11-25:
node js版本太低的缘故
希望OUI的作者能优化一下
这个检测node js版本的功能放到make阶段
而不是make menuconfig阶段
弹这个东西真的很烦

yuzhe1129, 2022-11-25:
> 

node JS是什么啊？那怎么解决这个问题

rongcenter, 2022-11-25:
升级一nodejs版本好了，14.18+

yuzhe1129, 2022-11-26:
> 

> 升级一nodejs版本好了，14.18+

怎么升级？不会呀

zheshifandian, 2022-11-26:
logs/package/feeds/oui/oui-app-acl/dump.txt
打开这文件就知道什么原因了

closed 2022-12-03 by aiamadeus