coolsnowwolf/lede #10500: ubnt ER-X 无法驱动eth0网口

lwb1978, 2022-11-26:
半年之前用lede编译的er-x固件都正常工作，今天用新源码重新编译了一下，更新后eth0网口无法驱动，网卡指示灯都不亮，看日志发现有一条错误是：mtk_soc_eth 1e100000.ethernet eth0: error -17 registering interface eth0

编译的内核版本是：5.4.224

maxmib, 2022-11-27:
相同问题

jaynfs, 2023-02-19:
r23.2.14仍有问题，暂时把eth1当wan了

zhang-liwei, 2023-05-01:
相同问题，用作lan wan都不行
[    0.000000] Linux version 5.4.241 (zzz@zzz-Ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r5911-702087ab3)) #0 SMP Sat Apr 22 15:54:32 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Ubiquiti EdgeRouter X
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
[    0.000000] pcpu-alloc: s26640 r8192 d22512 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250024K/262144K available (6211K kernel code, 271K rwdata, 1304K rodata, 1288K init, 235K bss, 12120K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015561] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.087872] pid_max: default: 32768 minimum: 301
[    0.097203] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.111605] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.129876] rcu: Hierarchical SRCU implementation.
[    0.140106] smp: Bringing up secondary CPUs ...
[    6.776335] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.776344] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.776356] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.776454] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.209590] Synchronize counters for CPU 1: done.
[    6.899495] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.899504] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.899513] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.899570] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.330182] Synchronize counters for CPU 2: done.
[    7.017006] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    7.017014] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    7.017023] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    7.017082] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.449773] Synchronize counters for CPU 3: done.
[    0.509381] smp: Brought up 1 node, 4 CPUs
[    0.521926] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.541399] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.555222] pinctrl core: initialized pinctrl subsystem
[    0.567400] NET: Registered protocol family 16
[    0.587532] FPU Affinity set after 11720 emulations
[    0.605072] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.625949] clocksource: Switched to clocksource GIC
[... 185 lines cut ...]
[   63.571539] br-lan: port 3(eth3) entered blocking state
[   63.582042] br-lan: port 3(eth3) entered forwarding state
[   64.596300] mt7530 mdio-bus:1f eth3: Link is Down
[   64.605890] br-lan: port 3(eth3) entered disabled state
[   67.716371] mt7530 mdio-bus:1f eth3: Link is Up - 1Gbps/Full - flow control rx/tx
[   67.731334] br-lan: port 3(eth3) entered blocking state
[   67.741751] br-lan: port 3(eth3) entered forwarding state
[  102.836614] mt7530 mdio-bus:1f eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[  102.852108] br-lan: port 1(eth1) entered blocking state
[  102.862651] br-lan: port 1(eth1) entered forwarding state
[  104.916805] mt7530 mdio-bus:1f eth1: Link is Down
[  104.926754] br-lan: port 1(eth1) entered disabled state
[... cut ...]
[ 1099.236563] mt7530 mdio-bus:1f eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 1099.251531] br-lan: port 1(eth1) entered blocking state
[ 1099.261952] br-lan: port 1(eth1) entered forwarding state
[ 1143.956327] mt7530 mdio-bus:1f eth1: Link is Down
[ 1143.965814] br-lan: port 1(eth1) entered disabled state
[ 1168.916567] mt7530 mdio-bus:1f eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[ 1168.931880] br-lan: port 1(eth1) entered blocking state
[ 1168.942300] br-lan: port 1(eth1) entered forwarding state

lwb1978, 2023-05-01:
都散了吧，ubnt-x应该是不会修复了的，我已经刷回了原版系统丢抽屉当备用交换机了。

skydevil88, 2024-02-18:
有人解决了吗？

coolsnowwolf, 2026-09-05:
该问题来自 2022 年 MT7621 旧版 5.4.224 树。当前 LEDE 已持续修复 EdgeRouter X 支持：现有 DTS、DSA 网络初始化、eth0/交换机端口映射和专用升级脚本均已包含，且后续提交还修正了 EdgeRouter X/SFP 的 RGMII 引脚声明与镜像尺寸。旧日志中的 `mtk_soc_eth ... error -17 registering interface eth0` 已不对应当前实现，无法在现行树复现；请使用当前 `ramips/mt7621` 镜像测试。issue 关闭。

closed 2026-09-05 by coolsnowwolf