coolsnowwolf/lede #10501: 新源码编译出错，路由器是N1及HK1等，日志如下

liuhejun2020, 2022-11-26:
from ./include/linux/memremap.h:5,
                 from ./include/linux/mm.h:29,
                 from /home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/cryptodev-linux-cryptodev-linux-1.12/ioctl.c:37:
./include/linux/irqreturn.h:17:21: error: missing terminating > character
   17 | #include <asm/unalig
      |                     ^
./include/linux/irqreturn.h:17:10: fatal error: asm/unalig: No such file or directory
   17 | #include <asm/unalig
      |          ^
cc1: some warnings being treated as errors
compilation terminated.
make[6]: *** [scripts/Makefile.build:286: /home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/cryptodev-linux-cryptodev-linux-1.12/ioctl.o] Error 1
make[5]: *** [Makefile:1837: /home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/cryptodev-linux-cryptodev-linux-1.12] Error 2
make[5]: Leaving directory '/home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.155'
make[4]: *** [Makefile:27: build] Error 2
make[4]: Leaving directory '/home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/cryptodev-linux-cryptodev-linux-1.12'
make[3]: *** [Makefile:61: /home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/cryptodev-linux-cryptodev-linux-1.12/.built] Error 2
make[3]: Leaving directory '/home/lhj/lede/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#65.56#1.44#294.64
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [package/Makefile:110: /home/lhj/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:231：world] 错误 2

free-diy, 2022-11-26:
x86也报这个错

crecywang, 2022-11-28:
make[1]: *** [package/Makefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/root/lede'
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:231: world] Error 2

closed 2023-01-14 by aiamadeus