coolsnowwolf/lede #10504: X86 一直编译不过同

selfcan, 2022-11-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Unhandled python exception

### (2) 路由器型号和固件版本

A：X6

### (3) 详细日志

A： File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 2554, in run
    super().run()
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 148, in run
    self.evaluate_codeblock(self.ast, start=1)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 173, in evaluate_codeblock
    raise e
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 166, in evaluate_codeblock
    self.evaluate_statement(cur)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 181, in evaluate_statement
    self.assignment(cur)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 566, in assignment
    value = self.evaluate_statement(node.value)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 179, in evaluate_statement
    return self.function_call(cur)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 454, in function_call
    res = func(node, func_args, kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  [Previous line repeated 5 more times]
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 96, in wrapped
    ret = f(*wrapped_args, **wrapped_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 115, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 276, in wrapper
    return f(*nargs, **wrapped_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 1577, in func_dependency
    d = df.lookup(kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 347, in lookup
    dep = func(kwargs, func_args, func_kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 90, in _do_dependency
    dep = dependencies.find_external_dependency(name, self.environment, kwargs)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/detect.py", line 114, in find_external_dependency
    d = c()
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 94, in __init__
    self._set_cargs()
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 203, in _set_cargs
    ret, out, err = self._call_pkgbin(['--cflags', self.name], env=env)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 157, in _call_pkgbin
    cache[(self.pkgbin, targs, fenv)] = self._call_pkgbin_real(args, env)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 115, in _call_pkgbin_real
    p, out, err = Popen_safe(cmd, env=env)
  File "/home/deepin/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/mesonlib/universal.py", line 1397, in Popen_safe
    o, e = p.communicate(write)
  File "/usr/lib/python3.10/subprocess.py", line 1152, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/usr/lib/python3.10/subprocess.py", line 2041, in _communicate
    stdout = self._translate_newlines(stdout,
  File "/usr/lib/python3.10/subprocess.py", line 1029, in _translate_newlines
    data = data.decode(encoding, errors)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 16: invalid continuation byte

../../../../build_dir/target-mipsel_24kc_musl/glib-2.74.0/meson.build:2046:0: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
make[3]: *** [Makefile:125: /home/deepin/文档/lede/build_dir/target-mipsel_24kc_musl/glib-2.74.0/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 2
make[3]: Leaving directory '/home/deepin/文档/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#10.75#4.39#14.00
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/home/deepin/文档/lede'
make[1]: *** [package/Makefile:110: /home/deepin/文档/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/deepin/文档/lede'
make: *** [/home/deepin/文档/lede/include/toplevel.mk:231：world] 错误 2

EnnawYang, 2022-11-26:
换全英文路径试试？

Pantheon-Paradoxes, 2022-11-26:
一样的问题，只有等官方修复了

bootli, 2022-11-26:
是的 添加个passwall 就编译不过 什么都没添加就可以编译

closed 2022-11-27 by aiamadeus

happyzhang1995, 2022-12-30:
你是怎么解决这个问题的，我也报一样的错误