coolsnowwolf/lede #10507: luci-app-mosdns

gswangmd-gssi, 2022-11-27:
编译顺利通过，但是mosdns管理界面显示“未运行”。

starxu1971, 2022-11-27:
这个去作者那里问，还有重启看看

dsyo2008, 2022-11-27:
确实有这个问题，启用后显示未运行，发下日志
Sun Nov 27 12:24:53 2022 daemon.err mosdns[7700]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:24:53 2022 daemon.err mosdns[7700]: 2022-11-27T04:24:53.106Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:24:58 2022 daemon.err mosdns[7800]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:24:58 2022 daemon.err mosdns[7800]: 2022-11-27T04:24:58.327Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:03 2022 daemon.err mosdns[8065]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:03 2022 daemon.err mosdns[8065]: 2022-11-27T04:25:03.546Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:08 2022 daemon.err mosdns[8258]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:08 2022 daemon.err mosdns[8258]: 2022-11-27T04:25:08.772Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:14 2022 daemon.err mosdns[8624]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:14 2022 daemon.err mosdns[8624]: 2022-11-27T04:25:14.000Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:19 2022 daemon.err mosdns[8750]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:19 2022 daemon.err mosdns[8750]: 2022-11-27T04:25:19.227Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:25:19 2022 daemon.info procd: Instance mosdns::mosdns s in a crash loop 6 crashes, 0 seconds since last crash
Sun Nov 27 12:26:15 2022 daemon.err mosdns[9562]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:15 2022 daemon.err mosdns[9562]: 2022-11-27T04:26:15.909Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:21 2022 daemon.err mosdns[9643]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:21 2022 daemon.err mosdns[9643]: 2022-11-27T04:26:21.123Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:26 2022 daemon.err mosdns[9661]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:26 2022 daemon.err mosdns[9661]: 2022-11-27T04:26:26.360Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:31 2022 daemon.err mosdns[9673]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:31 2022 daemon.err mosdns[9673]: 2022-11-27T04:26:31.592Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:36 2022 daemon.err mosdns[9691]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:36 2022 daemon.err mosdns[9691]: 2022-11-27T04:26:36.821Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:42 2022 daemon.err mosdns[9744]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Sun Nov 27 12:26:42 2022 daemon.err mosdns[9744]: 2022-11-27T04:26:42.054Z	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist

pangfeng161, 2022-11-27:
使用没问题呀,自己设置问题

mjyhj, 2022-11-27:
ssh里 services mosdns start 看看
他这个/etc/init.d/mosdns 很多跑不了。

推荐直接使用https://github.com/pmkol/easymosdns
这个luci删掉得了

fonlan, 2022-11-28:
你试试用https://github.com/sbwml/v2ray-geodata的Makefile替换ssrp下的v2ray-geodata/Makefile

oldskooler, 2022-11-30:
楼主解决了吗，这个问题我也遇到了，总是启动不起来

wx1wxc, 2022-11-30:
不保留配置更新固件试试

starxu1971, 2022-11-30:
也有可能是geoip.dat和geosite.dat两个文件新版的有问题替换旧版的，路径在/usr/share/v2ray下面。

starxu1971, 2022-11-30:
> ssh里 services mosdns start 看看 他这个/etc/init.d/mosdns 很多跑不了。
> 
> 推荐直接使用https://github.com/pmkol/easymosdns 这个luci删掉得了

我用了这个后有些游戏网页打不开了，比如战舰世界的军火库，启动器的新闻，仓库等等，其他倒是很好用，可能有些列表没包含上述网站地址。

xkelvinx666, 2022-12-02:
ssh进设备，手动执行/usr/bin/mosdns start -c /etc/mosdns/config.yaml 看报错

如果是手动配置跑这个/usr/bin/mosdns start -c /etc/mosdns/config_custom.yaml

WYC-2020, 2022-12-02:
这个用内置配置就行啊，能运行工作正常，不过看了这个模式 和ps和clash 搭配的好好看下怎么设置，ssr到很好与之配合
[image]

oldskooler, 2022-12-07:
日志里面出现如下信息，应该是geosite文件出错，是不是和ssrplus有冲突啊
`Wed Dec  7 12:43:47 2022 daemon.err mosdns[11197]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:43:47 2022 daemon.err mosdns[11197]: 2022-12-07T12:43:47.013+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:43:52 2022 daemon.err mosdns[11289]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:43:52 2022 daemon.err mosdns[11289]: 2022-12-07T12:43:52.160+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:43:57 2022 daemon.err mosdns[11331]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:43:57 2022 daemon.err mosdns[11331]: 2022-12-07T12:43:57.303+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:02 2022 daemon.err mosdns[11426]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:02 2022 daemon.err mosdns[11426]: 2022-12-07T12:44:02.448+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:07 2022 daemon.err mosdns[11490]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:07 2022 daemon.err mosdns[11490]: 2022-12-07T12:44:07.592+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:12 2022 daemon.err mosdns[11555]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:12 2022 daemon.err mosdns[11555]: 2022-12-07T12:44:12.731+0800	fatal	mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn does not exist
Wed Dec  7 12:44:12 2022 daemon.info procd: Instance mosdns::mosdns s in a crash loop 6 crashes, 1 seconds since last crash`

xkelvinx666, 2022-12-07:
> 日志里面出现如下信息，应该是geosite文件出错，是不是和ssrplus有冲突啊 `Wed Dec 7 12:43:47 2022 daemon.err mosdns[11197]: Error: mosdns exited, failed to init plugin #9, failed to load data from provider geosite, tag apple-cn [...]

apple-cn是[Loyalsoldier](https://github.com/Loyalsoldier/v2ray-rules-dat)的geosite才有。默认的是不带的，你可以自定义规则里去掉apple-cn，或者手动替换geosite，ps: 注意留学软件的自动更新

coolsnowwolf, 2026-09-05:
mosdns 仅报告界面显示“未运行”，没有 mosdns init 日志、配置文件或进程状态；无法区分配置禁用、二进制缺失和运行失败，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf