coolsnowwolf/lede #10510: Samba4共享名称、文件目录输入框太小

fxxccp, 2022-11-27:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如图所示
[image]

### (2) 路由器型号和固件版本

A：x86，5.15.80

### (3) 详细日志

A：无

starxu1971, 2022-11-27:
这个和你的浏览器有关，和固件没关系。

Plutonium141, 2022-11-27:
也许是主题的问题 换一下主题看看？

Plutonium141, 2022-11-27:
刚看了一下 确实是luci-app-samba4的页面问题 bootstrap主题下页面会被拉宽
[image]

fxxccp, 2022-11-28:
> 刚看了一下 确实是luci-app-samba4的页面问题 bootstrap主题下页面会被拉宽 [image]

感谢解惑，的确是主题原因导致。看起来得向主题作者提交issues。

Plutonium141, 2022-11-28:
> 

啊？我是说这不是主题的问题，是这个插件界面做的不好 元素加的太多导致放不下

closed 2026-09-05 by coolsnowwolf