coolsnowwolf/lede #10511: 最新源码编译，aria2显示未运行

kxjhcmc, 2022-11-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
使用最新源码编译升级后，aria2显示未运行
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：Sun Nov 27 18:58:08 2022 daemon.info aria2: Aria2 will run with user 'root'.
Sun Nov 27 18:58:08 2022 daemon.info aria2c[4917]: jail: exec-ing /usr/bin/aria2c
Sun Nov 27 18:58:09 2022 daemon.err aria2c[4917]: Specify at least one URL.
Sun Nov 27 18:58:09 2022 daemon.err aria2c[4917]: Usage: aria2c [OPTIONS] [URI | MAGNET | TORRENT_FILE | METALINK_FILE]...
Sun Nov 27 18:58:09 2022 daemon.err aria2c[4917]: See 'aria2c -h'.
Sun Nov 27 18:58:09 2022 daemon.info aria2c[4917]: jail: jail (4919) exited with exit: 1
Sun Nov 27 18:58:09 2022 daemon.info procd: Instance aria2::aria2.main s in a crash loop 7 crashes, 0 seconds since last crash

gevikhn, 2022-11-27:
遇到同样的问题

starxu1971, 2022-11-29:
显示未运行，但后台是在运行的。

kxjhcmc, 2022-11-29:
> Refer
并没有哈，你看日志也能看出来，aria2进程崩掉了。

starxu1971, 2022-11-29:
> > 显示未运行，但后台是在运行的。
> 
> 并没有哈，你看日志也能看出来，aria2 进程崩掉了。

我的就是这样后台运行着。界面不显示运行状态。从客户端能连接AriaNgGUI

wzb261643358, 2022-12-31:
权限问题，授予相应权限，并且配置目录最好在根目录内

wzb261643358, 2022-12-31:
如果更改过默认用户名，请授予再相关用户名权限下

coolsnowwolf, 2026-09-05:
日志明确显示 aria2c 启动时没有任何 URL（`Specify at least one URL`），随后进入 crash loop；这是 aria2 配置缺少下载任务/URL，不是守护进程包崩溃，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf