coolsnowwolf/lede #10514: 新装的ubuntu 20.04 LTS 中 amule 编译错误

aaronbagot, 2022-11-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译环境 Debian GNU/Linux 11 (bullseye)  和 Ubuntu 20.04.5 LTS 都试了，但amule编译一直出错。
检查log发现是gettext不支持某个旧参数了。但是gettext早就升级到0.20以上了，以前编译也没错，不知道哪儿的问题。
复现错误的方法：新装的debian 或 Ubuntu 20.04 LTS :
sudo apt update -y
sudo apt full-upgrade -y
sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev

git clone https://github.com/coolsnowwolf/lede
cd lede
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
选上luci-app-amule, 其它任何地方都不动
然后
make download -j17
make -j17
出错后 复现log: make package/amule/compile -j1 V=s 

### (2) 路由器型号和固件版本

A：X86-64-generic R22.10.1

### (3) 详细日志

A：make[2]: Entering directory '/home/user1/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/user1/lede/scripts/config'
make[1]: Entering directory '/home/user1/lede'
make[2]: Entering directory '/home/user1/lede/package/libs/toolchain'
echo "libc" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/user1/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.09#0.01#0.10
make[2]: Entering directory '/home/user1/lede/package/libs/zlib'
echo "zlib" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/user1/lede/package/libs/zlib'
time: package/libs/zlib/compile#0.09#0.00#0.09
make[2]: Entering directory '/home/user1/lede/package/devel/binutils'
echo "libbfd" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
make[2]: Leaving directory '/home/user1/lede/package/devel/binutils'
time: package/devel/binutils/compile#0.23#0.03#0.22
make[2]: Entering directory '/home/user1/lede/feeds/packages/libs/libpng'
echo "libpng" >> /home/user1/lede/staging_dir/target-x86_64_musl/pkginfo/libpng.default.install
make[2]: Leaving directory '/home/user1/lede/feeds/packages/libs/libpng'
time: package/feeds/packages/libpng/compile#0.10#0.00#0.09
make[2]: Entering directory '/home/user1/lede/feeds/packages/libs/freetype'
make[2]: Leaving directory '/home/user1/lede/feeds/packages/libs/freetype'
time: package/feeds/packages/freetype/compile#0.08#0.02#0.09
make[2]: Entering directory '/home/user1/lede/feeds/packages/libs/libjpeg-turbo'
make[2]: Leaving directory '/home/user1/lede/feeds/packages/libs/libjpeg-turbo'
[... 218 lines cut ...]
configure: WARNING: unrecognized options: --without-x
configure: loading site script /home/user1/lede/include/site/x86_64
checking for a BSD-compatible install... /home/user1/lede/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking for x86_64-openwrt-linux-strip... x86_64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking if this is a FreeBSD 4 or earlier system... no
checking whether the C++ compiler works... no
configure: error: in `/home/user1/lede/build_dir/target-x86_64_musl/aMule-2.3.3':
configure: error: C++ compiler cannot create executables
See `config.log' for more details
make[2]: *** [Makefile:115: /home/user1/lede/build_dir/target-x86_64_musl/aMule-2.3.3/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[2]: Leaving directory '/home/user1/lede/feeds/packages/net/amule'
time: package/feeds/packages/amule/compile#4.39#0.24#4.53
    ERROR: package/feeds/packages/amule failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/amule/compile] Error 1
make[1]: Leaving directory '/home/user1/lede'
make: *** [/home/user1/lede/include/toplevel.mk:231: package/amule/compile] Error 2

bootli, 2022-11-27:
一样

zhg-em, 2022-11-28:
我也遇到同样的问题，希望能够修复！

WYC-2020, 2022-11-29:
去掉liconv

coolsnowwolf, 2022-11-29:
fixed

closed 2022-11-29 by coolsnowwolf

aaronbagot, 2022-11-29:
多谢，问题解决了。