coolsnowwolf/lede #10515: ERROR: please fix feeds/oui/applications/oui-app报错

netlovehf, 2022-11-28:
Already up to date.
Create index file './feeds/packages.index'
Create index file './feeds/luci.index'
Create index file './feeds/routing.index'
Create index file './feeds/telephony.index'
Create index file './feeds/oui.index'
ERROR: please fix feeds/oui/applications/oui-app-acl/Makefile - see logs/feeds/oui/applications/oui-app-acl/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-backup/Makefile - see logs/feeds/oui/applications/oui-app-backup/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-demo/Makefile - see logs/feeds/oui/applications/oui-app-demo/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-dhcp-lease/Makefile - see logs/feeds/oui/applications/oui-app-dhcp-lease/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-home/Makefile - see logs/feeds/oui/applications/oui-app-home/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-layout/Makefile - see logs/feeds/oui/applications/oui-app-layout/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-login/Makefile - see logs/feeds/oui/applications/oui-app-login/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-stations/Makefile - see logs/feeds/oui/applications/oui-app-stations/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-system/Makefile - see logs/feeds/oui/applications/oui-app-system/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-upgrade/Makefile - see logs/feeds/oui/applications/oui-app-upgrade/dump.txt for details
ERROR: please fix feeds/oui/applications/oui-app-user/Makefile - see logs/feeds/oui/applications/oui-app-user/dump.txt for details
ERROR: please fix feeds/oui/oui-ui-core/Makefile - see logs/feeds/oui/oui-ui-core/dump.txt for details
Collecting package info: done

ecrasy, 2022-11-28:
#10496

netlovehf, 2022-11-30:
> #10496

thank you

closed 2022-12-03 by aiamadeus