coolsnowwolf/lede #10517: 编译报错：Patch failed!  Please fix ./patches/020-5.18-support.patch!

devioa, 2022-11-28:
## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题
```
Applying ./patches/001-r8168-add-LED-configuration-from-OF.patch using plaintext: 
patching file src/r8168_n.c

Applying ./patches/020-5.18-support.patch using plaintext: 
patching file src/r8168_n.c
Hunk #1 FAILED at 3715.
Hunk #2 FAILED at 3743.
Hunk #3 FAILED at 26464.
3 out of 3 hunks FAILED -- saving rejects to file src/r8168_n.c.rej
Patch failed!  Please fix ./patches/020-5.18-support.patch!
make[3]: *** [Makefile:46: /home/devioa/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.051.02/.prepared_106caea1bcc7b915bc3f21074e542b67_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/devioa/lede/package/lean/r8168'
time: package/lean/r8168/compile#0.07#0.03#0.10
    ERROR: package/lean/r8168 failed to build.
make[2]: *** [package/Makefile:116: package/lean/r8168/compile] Error 1
make[2]: Leaving directory '/home/devioa/lede'
make[1]: *** [package/Makefile:110: /home/devioa/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/devioa/lede'
make: *** [/home/devioa/lede/include/toplevel.mk:231：world] 错误 2
```

### (2) 路由器型号和固件版本
`Linux OpenWrt 5.15.79 #0 SMP Fri Nov 25 07:26:02 2022 x86_64 GNU/Linux`

### (3) 详细日志
[image]

bi4tms, 2022-11-28:
6.0内核出现同样错误

1582130940, 2022-11-28:
把020-5.18-support.patch删了试试

closed 2022-11-29 by aiamadeus