coolsnowwolf/lede #10519: 最新固件开启广告屏蔽大师无法上网

siren202101, 2022-11-28:
最新固件开启广告屏蔽大师无法上网

shenyz, 2022-11-28:
刚刚编译 表示没有问题
[image]
[image]
[image]
[image]

starxu1971, 2022-11-28:
有问题开了无法上网。

firkerword, 2022-11-28:
> 有问题开了无法上网。

看一下系统日志

firkerword, 2022-11-28:
> 最新固件开启广告屏蔽大师无法上网

看一下系统日志

siren202101, 2022-11-28:
> 刚刚译 表示没有问题 [image] [image] [image] [image]

试试全局模式

nb12nb34, 2022-11-28:
物理机安装，确实是打不开网页了。。连不上网，主要是回退了。看不了日志，最近的几个版本都有这个问题。。

yanxin152133, 2022-11-28:
没啥用，卸载了吧，之前我就是因为这个插件断网的

rongcenter, 2022-11-28:
> > 最新固件开启广告屏蔽大师无法上网
> 
> 看一下系统日志

Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:27 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:27 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:28 2022 daemon.info procd: - init complete -
Sat Nov 19 22:52:32 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:32 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:32 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:32 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:37 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:37 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:37 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:37 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:42 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:42 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:42 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:42 2022 user.err : jail: failed to load dependencies
Sat Nov 19 22:52:47 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sat Nov 19 22:52:47 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Nov 19 22:52:47 2022 user.err : jail: ntpd is not an absolute path
Sat Nov 19 22:52:47 2022 user.err : jail: failed to load dependencies

基本都是这个问题，而且最近发现，更新成功后只要重启就会出现这个问题。

yanxin152133, 2022-11-28:
> Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory Sat Nov 19 22:52:27 2022 daemon.crit dnsmasq[1]: FAILED to start up Sat [...]

跟我以前一样的报错，这个广告屏蔽的插件好像有个东西失效了就会影响DHCP，你的设备没办法分配到ip，这个插件也没啥用就别编译了，我现在就编译这几个插件
[image]

yanxin152133, 2022-11-28:
https://github.com/coolsnowwolf/lede/issues/10485

firkerword, 2022-11-29:
> 

我查看目录有这个文件夹 文件也都在 打开看也正常 权限也正常 新固件刚刷上只联网启动adbyby，其他插件都没开，也报错缺少文件  困扰我半个月了

nb12nb34, 2022-11-29:
哪个大佬出来解决下？暂时就发现这个具体原因不知道。。

firkerword, 2022-11-29:
> 

> 哪个大佬出来解决下？暂时就发现这个具体原因不知道。。

我估计是源码之外的其他插件导致的  日志看不出来

jerichon, 2022-11-29:
同样的情况，打开屏蔽大师插件不能上网，关闭之后立即就能恢复网络

starxu1971, 2022-11-29:
下午编译的去掉屏蔽大师的固件可以上网，但是SSR等软件重启后连接的总比以前的固件慢。要过半分钟左右国内国外才能连接，以前保存应用完国内网站是秒开的。

Ginua8292, 2022-11-30:
反正最近的编译的固件各种问题！没了广告大师，passwll开了全部设备都上不了网！用回旧的固件一点问题都没有

firkerword, 2022-12-01:
> 应该是与dhcp有冲突，客户端固定ip地址就可以上网

dnsmasq用这个版本https://github.com/coolsnowwolf/lede/tree/81a3a0ff39290e874d9ef9c9db52d2dfae9f40ca/package/network/services/dnsmasq

starxu1971, 2022-12-02:
> 反正最近的编译的固件各种问题！没了广告大师，passwll 开了全部设备都上不了网！用回旧的固件一点问题都没有

还是需要重启 dnsmasq

slipdaly, 2022-12-02:
应该和dnsmasq有关，感觉回退没回退好，目前停留在10月21日编译的版本，稳定运行中

siren202101, 2022-12-07:
> 应该和dnsmasq有关，感觉回退没回退好，目前停留在10月21日编译的版本，稳定运行中

大佬，如何退回版本？

lyx32, 2022-12-08:
改成全局模式就可以了。

wozualong, 2022-12-09:
因为TTYD莫名消失，编译了个新版本出现同样问题，回退回双十一编译的版本了

markvlenvision, 2022-12-11:
这个问题修复了没？

rongcenter, 2022-12-11:
> 这个问题修复了没？

没有，不清楚是啥问题，有人清楚嘛

rongcenter, 2022-12-16:
大家快去用，l大修复了。但是要改个文件。

init.d/adbyby

WAN_FILE=/var/etc/dnsmasq-adbyby.d/03-adbyby-ipset.conf
改成
WAN_FILE=/tmp/dnsmasq.d/dnsmasq-adbyby.d/03-adbyby-ipset.conf

xiangfeigao, 2022-12-17:
不手动改还是不行是吗？手动改的话，这个变量在哪个文件里，说详细下

coolsnowwolf, 2026-09-05:
“广告屏蔽大师开启后无法上网”没有设备、DNS 日志、规则集或配置，无法区分 DNS/防火墙/规则误拦截，也没有当前版本复现证据，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf