coolsnowwolf/lede #10520: softether  l2tp vpn暂时解决办法

shenyz, 2022-11-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
x86

### (3) 详细日志
无
A：
新编译的softether，用l2tp拨号，默认监听的IP为V6导致拨号无响应，表现为
netstat -anp |grep vpn
:::4500
:::500
......

1，先配置好softethervpn
2，停止服务   ./etc/init.d/softethervpnserver stop
3，编辑 /usr/libexec/softethervpn/vpn_server.config 文件   在90行 把string ListenIP ：：： 改成 string ListenIP 0.0.0.0  保存
4，再启动服务  ./etc/init.d/softethervpnserver start   就可以连接了

netstat -anp |grep vpn
tcp        0      0 0.0.0.0:992             0.0.0.0:*               LISTEN      3198/vpnserver
tcp        0      0 10.88.188.1:992         10.88.188.96:63346      ESTABLISHED 3198/vpnserver
udp        0      0 0.0.0.0:1194            0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:58902           0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:4500            0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:5555            0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:56925           0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:443             0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:500             0.0.0.0:*                           3198/vpnserver
udp        0      0 0.0.0.0:992             0.0.0.0:*                           3198/vpnserver

WYC-2020, 2022-12-06:
这狗东西早就修复了，只是作者一直没有打包release,git仓库早就合并了，https://github.com/SoftEtherVPN/SoftEtherVPN/pull/1416

WYC-2020, 2022-12-06:
[image]
验证完毕 可以登录，自己fork那个仓库自己打一个tag,然后改下下载源，即可食用最新版，忘了说还得更新patch 并且编译host的时候的打一个patch就可以了，因为openwrt host编译的时候不是用的openssl，导致有函数找不到
[image]

coolsnowwolf, 2026-09-05:
SoftEther L2TP 监听地址修复属于用户生成的 vpn_server.config 配置调整，缺少当前版本复现与兼容性要求，不提交默认改变监听地址的补丁，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf