coolsnowwolf/lede #10521: 最新源码编译出错，本地和云编译都不能通过！

liuhejun2020, 2022-11-29:
X86 64

sconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --disable-static --disable-rpath --with-gnu-ld --disable-ccache --disable-optimize --disable-profile --disable-monolithic --enable-amule-daemon --enable-amulecmd --enable-webserver --disable-amule-gui --disable-cas --disable-wxcas --disable-ed2k --disable-alc --disable-alcc --disable-fileview --disable-plasmamule --without-wxdebug --enable-dlp --enable-upnp --with-zlib="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --with-gdlib-prefix="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --with-libpng-prefix="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --with-wx-prefix="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --with-crypto-prefix="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --with-libiconv-prefix="" --with-libintl-prefix="" --with-libupnp-prefix="/home/lhj/lede/staging_dir/target-x86_64_musl/usr" --without-x --disable-debug  ; fi; )
configure: WARNING: unrecognized options: --without-x
configure: loading site script /home/lhj/lede/include/site/x86_64
checking for a BSD-compatible install... /home/lhj/lede/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking for x86_64-openwrt-linux-strip... x86_64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking if this is a FreeBSD 4 or earlier system... no
checking whether the C++ compiler works... no
configure: error: in `/home/lhj/lede/build_dir/target-x86_64_musl/aMule-2.3.3':
configure: error: C++ compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:115: /home/lhj/lede/build_dir/target-x86_64_musl/aMule-2.3.3/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/home/lhj/lede/feeds/packages/net/amule'
time: package/feeds/packages/amule/compile#17.88#1.94#23.19
    ERROR: package/feeds/packages/amule failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/amule/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [package/Makefile:110: /home/lhj/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:231：world] 错误 2
lhj@debian:~/lede$

yanxin152133, 2022-11-29:
https://github.com/coolsnowwolf/lede/issues/10514

liuhejun2020, 2022-11-29:
我用的新装的Debian11.5

WYC-2020, 2022-11-29:
看另外一个帖子我有说

coolsnowwolf, 2022-11-29:
fixed

closed 2026-09-05 by coolsnowwolf