coolsnowwolf/lede #10524: 最新源码编译报错 歌华链 MT7621 GeHua GHL-R-001

TokenYan233, 2022-11-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
最新源码编译报错 MT7621 GeHua GHL-R-001

### (1) 具体问题

A：选择机型后只勾选了Build the OpenWrt Image Builder和Minify Lua sources就保存，一样报错

### (2) 路由器型号和固件版本

A：MT7621 GeHua GHL-R-001 歌华链

### (3) 详细日志

A：
make[6]: Leaving directory '/workdir/openwrt/target/linux/ramips/image/lzma-loader'
mv "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin.bin" "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin"
rm -rf /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin.src
mkimage -A mips -O linux -T kernel -C none -a 0x80001000 -e 0x80001000 -n 'MIPS OpenWrt Linux-5.4.225'   -d /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin.new
Image Name:   MIPS OpenWrt Linux-5.4.225
Created:      Mon Nov 28 17:10:37 2022
Image Type:   MIPS Linux Kernel Image (uncompressed)
Data Size:    2561813 Bytes = 2501.77 KiB = 2.44 MiB
Load Address: 80001000
Entry Point:  80001000
mv /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin.new /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/gnubee_gb-pc2-kernel.bin
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin
cp /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/vmlinux /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin
gzip -f -9n -c /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin  > /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.new
rm -fR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.initrd
rm -fR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.initrd.cpio
mkdir -p /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.initrd/{bin,dev,proc,sys,lib,etc}
mkdir -p /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.initrd/lib/modules/5.4.225
cp -fpR /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/rdloader/bin/* /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin.initrd/bin/
cp: cannot stat '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/rdloader/bin/*': No such file or directory
make[5]: *** [Makefile:217: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/hatlab_gateboard-one-kernel.bin] Error 1
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#250.57#14.61#261.14
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

Z139724, 2022-11-30:
我的也是MT7621   小米路由3pro和小米CR660X都报错
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:262: /home/zhang/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-11-24-daae6ca5/mt76_connac_mcu.o] Error 1
make[4]: *** [Makefile:1739: /home/zhang/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-11-24-daae6ca5] Error 2
make[4]: Leaving directory '/home/zhang/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.225'
make[3]: *** [Makefile:526: /home/zhang/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-11-24-daae6ca5/.built] Error 2
make[3]: Leaving directory '/home/zhang/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#1.09#0.32#1.32
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/zhang/lede'
make[1]: *** [package/Makefile:110: /home/zhang/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhang/lede'
make: *** [/home/zhang/lede/include/toplevel.mk:231：world] 错误 2

Z139724, 2022-11-30:
里面插件只留了luci-app-firewall #添加防火墙   一样还是这个报错....

yuzhe1129, 2022-11-30:
正好我也想编译这个固件，

commit 2173f1c0 (2022-11-30): mt76: fixes kernel 5.4 build error

Fixes: #10524
closed 2022-11-30 by aiamadeus