coolsnowwolf/lede #10525: 大神推荐用msd_lite代替udpxy

biankasyo, 2022-11-29:
项目链接：https://github.com/rozhuk-im/msd_lite

相对udpxy优势：

1、支持ipv6（配合DDNS，ipv6直通，实测无问题）
2、CPU和内存占用更低。据说msd只占用0.4%，udpxy占用4%（未对比）

xiangfeigao, 2022-11-30:
没有luci-app吧？

VergilGao, 2022-12-03:
> 没有luci-app吧？

简单看了一眼，配置挺简单的应该也用不上luci界面

moonerick, 2022-12-04:
这么多年了，总算有人打算来更新这个东西了，目前家里的iptv会48小时强制重连，而udpxy无法自动更新重连后的ip，不得已自己写了一个脚本来监控iptv的ip变动，而且看超高清频道，udpxy也会出现偶发性的假死

GhqZx, 2022-12-05:
已经有人做了luci-app了 
https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite
我自己编译了下 可以使用 性能比udpxy确实强了非常多 但是msd_lite和udpxy一样 无法自动更新ip 需要添加脚本重启

coolsnowwolf, 2022-12-09:
可以写个 pppoe 的 hotplug 脚本

VergilGao, 2022-12-10:
> 可以写个 pppoe 的 hotplug 脚本

对哦，openwrt有 hotplug 
拨号状态更新直接触发 hotplug 此时重启 udpxy 就行了

tanst, 2022-12-11:
没有 status 吗？

MageDelfador, 2023-01-17:
我试了下，用VLC播放的时候声音画面极其卡顿，但udpxy不会
大家都没遇到这个问题吗
我用的默认配置参数，只改了下那个ifname,改成pppoe-IPTV

biankasyo, 2023-01-20:
> 我试了下，用VLC播放的时候声音画面极其卡顿，但udpxy不会 大家都没遇到这个问题吗 我用的默认配置参数，只改了下那个ifname,改成pppoe-IPTV

BindToCPU改成yes（将线程绑定到CPU内核）
threadsCountMax改成对应路由器线程数
我这边修改后画面不会再出现撕裂情况，但某些频道间隔十几秒会轻微卡一下

MageDelfador, 2023-01-20:
是msd_lite的bug，我刚刚提交了PR
https://github.com/rozhuk-im/msd_lite/pull/5

MageDelfador, 2023-01-26:
> > 是msd_lite的bug，我刚刚提交了PR [rozhuk-im/msd_lite#5](https://github.com/rozhuk-im/msd_lite/pull/5)
> 
> 有没有露西？？只支持r19的源码，不支持r23

LUCI链接在楼上，r23没问题

MageDelfador, 2023-01-26:
> > > > 是msd_lite的bug，我刚刚提交了PR [rozhuk-im/msd_lite#5](https://github.com/rozhuk-im/msd_lite/pull/5)
> > > 
> > > 
> > > 有没有露西？？只支持r19的源码，不支持r23
> > 
> > 
> > LUCI链接在楼上，r23没问题
> 
> 我用了，不行

[image]
用的楼上链接，r23

MageDelfador, 2023-01-26:
试试求助下大佬@coolsnowwolf做个luci
最好顺便加个hotplug脚本功能

coolsnowwolf, 2023-01-30:
[image]

coolsnowwolf, 2023-01-30:
[image]

Nothend, 2023-02-21:
> [image]
太厉害了楼主，可是最新的编译里面，怎么找不到这个luci app呢？

Nothend, 2023-02-21:
> > > [image]
> > > 太厉害了楼主，可是最新的编译里面，怎么找不到这个luci app呢？
> 
> 因为本来就没有

额，那太可惜了，看来是作者自用了，能用到最好了，确实看上去比udpxy强太多了

Nothend, 2023-02-21:
> > > > > [image]
> > > > > 太厉害了楼主，可是最新的编译里面，怎么找不到这个luci app呢？
> > > 
> > > 
> > > 因为本来就没有
> > 
> > 
> > 额，那太可惜了，看来是作者自用了，能用到最好了，确实看上去比udpxy强太多了
> 
> 那有，lean不想加😂

哦 好的  谢谢，那只能尊重lean的意愿了，期待有一天能放出来~

moonerick, 2023-02-21:
> 

建议自己从https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite这里下载来进行修改，需要修改的地方很少，如果只做伸手党那就爱莫能助了。
目前我自己的能正常运行，没有发现任何问题，另外lean大佬的packages里面是有msd_lite的，但是我测试了一下，配合上面的那个luci app无法启动，什么原因我不太清楚，而且msd_lite原作者的软件包最后更新日期是2023-2-17日，而大佬的包里面还是2022-04-17这个版本

Nothend, 2023-02-21:
> https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite

谢谢了，不是伸手党，我来研究下，感谢感谢

moonerick, 2023-02-21:
> > https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite
> 
> 谢谢了，不是伸手党，我来研究下，感谢感谢

ip变化需要配合hotplug脚本，算了

> > https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite
> 
> 谢谢了，不是伸手党，我来研究下，就是不知道能不能监测IP的变化，感谢感谢

ip变化需要配合hotplug脚本来控制

上面那个作者的包里面包含了luci-app和msd_lite的内容，下载后将luci-app-msd_lite文件夹拷贝到lede\feeds\luci\applications
将剩下的msd_lite文件夹内容拷贝到lede\feeds\packages\net进行替换
然后修改msd_lite文件夹内的Makefile文件
我的有改动的地方如下：
PKG_VERSION:=1.40

PKG_SOURCE_DATE:=2023-02-17
PKG_SOURCE_VERSION:=744d2ef91797471e26b3b117e7aa0ffebbb91106
PKG_HASH:=b725462b1a8db6f2a0cce75eb426da7be5aa72f2a679290292bd1b6a941c2df0

Nothend, 2023-02-21:
> ip变化需要配合hotplug脚本来控制
>
> 上面那个作者的包里面包含了luci-app和msd_lite的内容，下载后将luci-app-msd_lite文件夹拷贝到lede\feeds\luci\applications 将剩下的msd_lite文件夹内容拷贝到lede\feeds\packages\net进行替换 然后修改msd_lite文件夹内的Makefile文件 [...]

太棒了，可以说很详细了！好人一生平安~

moonerick, 2023-02-21:
> 太棒了，可以说很详细了！好人一生平安~

监测ip变化的脚本就更简单了
写一个脚本如下：

#!/bin/bash
if [ $ACTION=ifup -a $INTERFACE=iptv ]; then
  sh /etc/init.d/msd_lite restart
fi

上面的iptv改成你的iptv接口名字
将这个脚本取个名字形如:95-msd_lite，放到/etc/hotplug.d/iface/ 文件夹下面，然后iptv这个wan口的状态变化后，系统就会自动重启msd_lite了

Nothend, 2023-02-21:
> > 太棒了，可以说很详细了！好人一生平安~
> 
> 监测ip变化的脚本就更简单了 写一个脚本如下：
> 
> #!/bin/bash if [ $ACTION=ifup -a $INTERFACE=iptv ]; then sh /etc/init.d/msd_lite restart fi
> 
> 上面的iptv改成你的iptv接口名字 将这个脚本取个名字形如:95-msd_lite，放到/etc/hotplug.d/iface/ 文件夹下面，然后iptv这个wan口的状态变化后，系统就会自动重启msd_lite了

你这。。保姆级的教程了。。祝大佬硬邦邦

OrdinaryJoys, 2023-02-23:
实测了msd，内网用确实减少资源占用了，但是他没有设置msd外网访问端口的功能和映射功能，替换msd后外网访问端口完全失效

termzh, 2023-03-26:
无法实现外网访问，端口映射无法使用

windybird, 2023-06-07:
新三路由，命令运行后只要打开播放路由器就死机了（lean自带msd_lite编译，版本
1.11-1），udpxy就没问题。。。就调了一个iframe项

Luo2018, 2023-07-03:
> > 
> 
> 
> 
> 建议自己从https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite这里下载来进行修改，需要修改的地方很少，如果只做伸手党那就爱莫能助了。
> 
> 目前我自己的能正常运行，没有发现任何问题，另外lean大佬的packages里面是有msd_lite的，但是我测试了一下，配合上面的那个luci app无法启动，什么原因我不太清楚，而且msd_lite原作者的软件包最后更新日期是2023-2-17日，而大佬的包里面还是2022-04-17这个版本

能加个好友吗？

Luo2018, 2023-07-13:
比udpxy好用吗？

windybird, 2023-07-15:
> 叫lea

外网映射不能用

Luo2018, 2023-08-05:
没更新啊

justsee233, 2023-09-22:
> 实测了msd，内网用确实减少资源占用了，但是他没有设置msd外网访问端口的功能和映射功能，替换msd后外网访问端口完全失效

可以的，只是内外网端口号需要设置成一样的

Luo2018, 2023-10-01:
哦

YK-Samgo, 2024-02-08:
借地说一下`msd_lite`使用过程中遇到的问题

上海电信IPTV每4分20秒会断一下，最终在[27楼](https://www.newsmth.net/nForum/#!article/DigiHome/944514?p=3)查到是`renew`没发出去，`msd_lite`中对应的配置项叫`rejoin_time`、中文`IGMP/MLD 重新加入时间`，在`luci-app`中默认值是`0`也就是不更新，我改成200就好了。

Nothend, 2024-02-09:
> 借地说一下`msd_lite`使用过程中遇到的问题
> 
> 上海电信IPTV每4分20秒会断一下，最终在[27楼](https://www.newsmth.net/nForum/#!article/DigiHome/944514?p=3)查到是`renew`没发出去，`msd_lite`中对应的配置项叫`rejoin_time`、中文`IGMP/MLD 重新加入时间`，在`luci-app`中默认值是`0`也就是不更新，我改成200就好了。

感谢，目前我倒没遇到你这个问题。江苏电信。插个眼

jimliu111, 2024-03-07:
renew和rejoin配置不是必须的，断流的原因是默认的防火墙规则只请允许出站的igmp请求，不允许入站的igmp主动查询，只要添加允许igmp入站规则，保证路由器正常响应igmp查询即可

YK-Samgo, 2024-03-13:
> 在：“防火墙-通信规则”里新增规则，选项：ipv4、协议igmp、源区域iptv（根据自定义实际区域确认）、目的选择此设备，允许入站

试了一下，防火墙区域IPTV85全接受、通信规则协议IGMP源区域IPTV85目标区域设备（输入），依然会发生4：20断一下，路由器也重启过了

xmtomcom, 2024-03-18:
> > 太棒了，可以说很详细了！好人一生平安~
> 
> 监测ip变化的脚本就更简单了 写一个脚本如下：
> 
> #!/bin/bash if [ $ACTION=ifup -a $INTERFACE=iptv ]; then sh /etc/init.d/msd_lite restart fi
> 
> 上面的iptv改成你的iptv接口名字 将这个脚本取个名字形如:95-msd_lite，放到/etc/hotplug.d/iface/ 文件夹下面，然后iptv这个wan口的状态变化后，系统就会自动重启msd_lite了

请问，这个脚本文件名要带后缀.js吗，小白恳请直接发一个脚本文件下载吧，谢谢了！

moonerick, 2024-03-18:
> > > 太棒了，可以说很详细了！好人一生平安~
> > 
> > 
> > 监测ip变化的脚本就更简单了 写一个脚本如下：
> > #!/bin/bash if [ $ACTION=ifup -a $INTERFACE=iptv ]; then sh /etc/init.d/msd_lite restart fi
> > 上面的iptv改成你的iptv接口名字 将这个脚本取个名字形如:95-msd_lite，放到/etc/hotplug.d/iface/ 文件夹下面，然后iptv这个wan口的状态变化后，系统就会自动重启msd_lite了
> 
> 请问，这个脚本文件名要带后缀.js吗，小白恳请直接发一个脚本文件下载吧，谢谢了！

不需要后缀扩展名

xmtomcom, 2024-03-19:
> 不需要后缀扩展名

就是直接用记事本新建一个没有后缀的文件，把这段脚本copy进去就可以了吗？

lqzhj, 2024-04-03:
大佬能

> 建议自己从[https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite这里下载来进行修改，需要修改的地方很少，如果只做伸手党那就爱莫能助了。](https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/ [...]

大佬能出个luci吗，lean大雕msd_lite更新到2024-03-15后 上面修改的luci app不能用了。

Gigasnow, 2024-08-25:
各位有遇到msd_lite不能被反向代理的情况吗？想加个https放到外网

Gigasnow, 2024-08-26:
> 各位有遇到msd_lite不能被反向代理的情况吗？想加个https放到外网

搞定了，把return 403的if判断条件里的HTTP_SRV_RD_F_HOST_IS_LOCAL去掉，重新编译ipk就好了

phuai007, 2024-09-13:
我用上了，出现一个问题，就是，重启路由后，2-4天不定时就不能看了，这时候进路由，重启msd_lite服务，重连iptv口，重连lan口都不行，只能重启路由才能恢复，不知道啥问题，翻遍网络都不行。

guolongchu, 2024-12-10:
> > 各位有遇到msd_lite不能被反向代理的情况吗？想加个https放到外网
> 
> 搞定了，把return 403的if判断条件里的HTTP_SRV_RD_F_HOST_IS_LOCAL去掉，重新编译ipk就好了

请问你说的这个修改的位置在何处，我翻了源码没找到！改动此处能解决端口映射到外网吗？

guolongchu, 2024-12-10:
> > 实测了msd，内网用确实减少资源占用了，但是他没有设置msd外网访问端口的功能和映射功能，替换msd后外网访问端口完全失效
> 
> 可以的，只是内外网端口号需要设置成一样的

终于解决了，在路由器的防火墙中放行msd_lite的端口，-----在防火墙中添加一条规则，将端口指向路由器，enjoy搞定

guolongchu, 2024-12-10:
> > > 太棒了，可以说很详细了！好人一生平安~
> > 
> > 
> > 监测ip变化的脚本就更简单了 写一个脚本如下：
> > #!/bin/bash if [ $ACTION=ifup -a $INTERFACE=iptv ]; then sh /etc/init.d/msd_lite restart fi
> > 上面的iptv改成你的iptv接口名字 将这个脚本取个名字形如:95-msd_lite，放到/etc/hotplug.d/iface/ 文件夹下面，然后iptv这个wan口的状态变化后，系统就会自动重启msd_lite了
> 
> 请问，这个脚本文件名要带后缀.js吗，小白恳请直接发一个脚本文件下载吧，谢谢了！

你可以用winscp连接到路由器的那个目录下，拷贝一个文件出来，把脚本复制进去，改一下IPTV的接口名称，然后重命名一下，上传进去就好了

wzyou, 2025-01-08:
msd_lite 支持ipv6 能直接访问确实很方便，但是msd_lite没有鉴权功能。大家都是怎么保证安全的

guolongchu, 2025-01-09:
> msd_lite 支持ipv6 能直接访问确实很方便，但是msd_lite没有鉴权功能。大家都是怎么保证安全的
防火墙你对指定ip放行，

Leoyzen, 2025-02-07:
> 各位有遇到msd_lite不能被反向代理的情况吗？想加个https放到外网

深入后发现是 msd_lite 会校验传入http 请求的 Host Header, 是否和监听IP及端口一致。

解决办法有两个：
1. 反代服务器时不传入`Host` Header。例如 traefik 中配置`passHostHeader: false` 即可，会忽略掉 host header。
2. 在`msd_lite`配置文件中，将域名加入 <hostname> 中。但是需要注意，监听端口也要一致(代码中默认是 80 端口)

所以建议还是第一个方案，直接忽略掉这个 header 即可。

coolsnowwolf, 2026-09-05:
该建议在当前 packages feed 中已有实现：net/msd_lite 包已存在，当前源码日期为 2025-05-02，版本提交为 79a6c62c8fced6128a5e445ee110709b3b51bb78，并通过 CMake 构建安装 /usr/bin/msd_lite 与 /etc/msd_lite 配置。issue 讨论的 LuCI 前端和 IPTV hotplug 脚本属于额外的第三方配置需求，不是该包的缺陷；现有 udpxy/msd_lite 均保留，用户可按接口变化自行添加 hotplug。按“功能已在 feed 提供”关闭此功能建议。

closed 2026-09-05 by coolsnowwolf