coolsnowwolf/lede #10529: 近期的源码正常编译后 上机都不能启动

Travel2Here, 2022-11-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：正常编译后 开机启动超过5分钟依然闪烁SYS灯，不能正常进入系统；
编译时配置内核容量64M，ROOT 900M，远超默认；
是否最近的码没有对R4S做过测试？还是我添加的其他app有冲突？（但跟过去正常启动的版本，在软件配置上没有区别）
请大神们指路。已经编了超过30次不同的版本了。。。。包括更换3台不同的设备，各种clean，全新；

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志

A： 
……
make[3] -C package/lean/default-settings compile
 make[2] package/install
 make[2] target/install
 make[3] -C target/linux install
 make[2] package/index
 make[2] json_overview_image_info
 make[2] checksum

sdf8057, 2022-11-30:
泪目

Travel2Here, 2022-12-01:
> 泪目

咋整 到底我的问题还是咱们代码问题？我已经哭干了

sdf8057, 2022-12-01:
https://github.com/immortalwrt/immortalwrt，用这个，这个稳

Travel2Here, 2022-12-01:
> https://github.com/immortalwrt/immortalwrt，用这个，这个稳

哈哈 我试试，兄弟这是属于挖墙脚吗😆

BoBppy, 2022-12-06:
我好像也是，r66s

closed 2022-12-06 by coolsnowwolf

keeno1982, 2022-12-19:
我也是，上周用最新的包编译后R4S无法启动，不知道后面有没有修复