coolsnowwolf/lede #10532: 使用VPS编译R66s固件时，弹出一个选择提示，要手动输入 yes

chika0801, 2022-11-30:
使用VPS编辑R66s固件时，弹出选项是

```
make[4]: Entering directory '/home/chika/openwrt/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2022.07'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
.config:2030:warning: override: reassigning to symbol IDENT_STRING
*
* Restart config...
*
*
* ARM architecture
*
Enable support for CRC32 instruction (ARM64_CRC32) [Y/n/?] y
Timer clock frequency (COUNTER_FREQUENCY) [0] 0
Generate position-independent pre-relocation code (POSITION_INDEPENDENT) [N/y/?] n
Specify the early stack pointer relative to the .bss section (INIT_SP_RELATIVE) [N/y/?] n
ARM GICV3 ITS (GIC_V3_ITS) [N/y/?] n
Do not enable icache (SYS_ICACHE_OFF) [N/y/?] n
Do not enable icache in SPL (SPL_SYS_ICACHE_OFF) [N/y/?] n
Do not enable dcache (SYS_DCACHE_OFF) [N/y/?] n
Do not enable dcache in SPL (SPL_SYS_DCACHE_OFF) [N/y/?] n
CP15 based cache enabling support (SYS_ARM_CACHE_CP15) [N/y/?] n
MMU-based Paged Memory Management Support (SYS_ARM_MMU) [N/y/?] n
Use the ARM v7 PMSA Compliant MPU (SYS_ARM_MPU) [N/y/?] n
Select the ARM data write cache policy
> 1. Write-back (WB) (SYS_ARM_CACHE_WRITEBACK)
  2. Write-through (WT) (SYS_ARM_CACHE_WRITETHROUGH)
  3. Write allocation (WA) (SYS_ARM_CACHE_WRITEALLOC)
choice[1-3?]: 1
Enable ARCH_CPU_INIT (ARCH_CPU_INIT) [N/y/?] n
ARM Generic Timer support (SYS_ARCH_TIMER) [Y/n/?] y
Support for ARM SMC Calling Convention (SMCCC) (ARM_SMCCC) [Y/?] y
Support ARM semihosting (SEMIHOSTING) [N/y/?] n
Support ARM semihosting in SPL (SPL_SEMIHOSTING) [N/y/?] n
L2cache off (SYS_L2CACHE_OFF) [N/y/?] n
prepare BOOT0 header (ENABLE_ARM_SOC_BOOT0_HOOK) [Y/?] y
Use an assembly optimized implementation of memcpy (USE_ARCH_MEMCPY) [N/y/?] (NEW) 

Use an assembly optimized implementation of memcpy for SPL (SPL_USE_ARCH_MEMCPY) [N/y/?] n

Use an assembly optimized implementation of memset (USE_ARCH_MEMSET) [N/y/?] (NEW) 

```

chika0801, 2022-11-30:
Use an assembly optimized implementation of memcpy (USE_ARCH_MEMCPY) [N/y/?] (NEW)  

Use an assembly optimized implementation of memset (USE_ARCH_MEMSET) [N/y/?] (NEW) 

这2个项目没有默认选项值，请问默认值是 yes 或 no 了？想让他编译时不人工选，自动完成

chika0801, 2022-12-01:
使用github在线编译时，是直接完成。

问题出在用VPS编译时。

coolsnowwolf, 2022-12-04:
depends on !ARM64 , 默认都是no

chika0801, 2022-12-04:
> depends on !ARM64 , 默认都是no

谢谢你的帮助，我是发现用VPS在线编译，这2个缺少了默认值no吧。

closed 2022-12-04 by chika0801