coolsnowwolf/lede #10533: R66s长时间满速下载，网卡掉线断流

AutoCONFIG, 2022-11-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R66s长时间满速下载，网卡掉线断流。
网卡协商速度为100M，满速下载一段时间后，会出现掉网卡的问题，之后pppoe会拨号失败，必须手动重播才能恢复正常
掉线情况挺频发，只要是满速下载一个小时内，必掉线至少一次。
网卡驱动为r8125
### (2) 路由器型号和固件版本

A：FastRhino R66S
OpenWrt R22.11.11 / LuCI Master (git-22.330.64765-00290b6) 
### (3) 详细日志

A：

Wed Nov 30 18:32:35 2022 kern.info kernel: [19955.210058] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:34:29 2022 kern.warn kernel: [20069.501039] net_ratelimit: 1605 callbacks suppressed
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.501536] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.503009] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.503400] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.505016] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.506043] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.509434] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.510598] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.511495] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.512536] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:29 2022 kern.info kernel: [20069.514360] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.warn kernel: [20075.145353] net_ratelimit: 2288 callbacks suppressed
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.145843] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.147153] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.148378] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.149700] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.150189] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.150737] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.151299] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.152527] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.155332] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.155798] neighbour: ndisc_cache: neighbor table overflow!
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.300432] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.301184] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:34:35 2022 kern.info kernel: [20075.301736] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:34:52 2022 daemon.err uhttpd[1277]: cat: can't open '/etc/bench.log': No such file or directory
Wed Nov 30 18:34:53 2022 daemon.err uhttpd[1277]: cat: can't open '/etc/bench.log': No such file or directory
Wed Nov 30 18:35:03 2022 daemon.err uhttpd[1277]: cat: can't open '/etc/bench.log': No such file or directory
Wed Nov 30 18:35:30 2022 daemon.err uhttpd[1277]: cat: can't open '/etc/bench.log': No such file or directory
Wed Nov 30 18:36:35 2022 kern.info kernel: [20195.384743] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:36:35 2022 kern.info kernel: [20195.385630] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:36:35 2022 kern.info kernel: [20195.386222] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:38:39 2022 kern.info kernel: [20319.481139] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:38:39 2022 kern.info kernel: [20319.481942] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:38:39 2022 kern.info kernel: [20319.482488] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:40:39 2022 kern.info kernel: [20439.582588] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:40:39 2022 kern.info kernel: [20439.583530] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:40:39 2022 kern.info kernel: [20439.584108] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:42:40 2022 kern.info kernel: [20559.661902] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:42:40 2022 kern.info kernel: [20559.662947] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:42:40 2022 kern.info kernel: [20559.663543] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:44:40 2022 kern.info kernel: [20679.765310] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:44:40 2022 kern.info kernel: [20679.766275] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:44:40 2022 kern.info kernel: [20679.766835] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:46:45 2022 kern.info kernel: [20805.153936] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 18:46:45 2022 kern.info kernel: [20805.154797] r8125 0001:01:00.0 eth0: rss get rxnfc
[... 376 lines cut ...]
Wed Nov 30 20:30:51 2022 daemon.notice pppd[11278]: remote LL address fe80::7244:00ff:fe4f:ff04
Wed Nov 30 20:30:51 2022 daemon.info avahi-daemon[1336]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::8ef6:6abe:f464:ef5c.
Wed Nov 30 20:30:51 2022 daemon.info avahi-daemon[1336]: New relevant interface pppoe-wan.IPv6 for mDNS.
Wed Nov 30 20:30:51 2022 daemon.info avahi-daemon[1336]: Registering new address record for fe80::8ef6:6abe:f464:ef5c on pppoe-wan.*.
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Interface 'wan' is now up
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Wed Nov 30 20:30:51 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Wed Nov 30 20:30:51 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Wed Nov 30 20:30:51 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Wed Nov 30 20:30:54 2022 daemon.info avahi-daemon[1336]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::8ef6:6abe:f464:ef5c.
Wed Nov 30 20:30:54 2022 daemon.info avahi-daemon[1336]: Joining mDNS multicast group on interface pppoe-wan.IPv6 with address 2409:8a34:2007:3dc:8ef6:6abe:f464:ef5c.
Wed Nov 30 20:30:54 2022 daemon.info avahi-daemon[1336]: Registering new address record for 2409:8a34:2007:3dc:8ef6:6abe:f464:ef5c on pppoe-wan.*.
Wed Nov 30 20:30:54 2022 daemon.info avahi-daemon[1336]: Withdrawing address record for fe80::8ef6:6abe:f464:ef5c on pppoe-wan.
Wed Nov 30 20:30:54 2022 daemon.info avahi-daemon[1336]: Registering new address record for 2409:8a34:2074:af30::1 on eth0.*.
Wed Nov 30 20:30:54 2022 daemon.notice netifd: Interface 'wan_6' is now up
Wed Nov 30 20:30:54 2022 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Wed Nov 30 20:31:07 2022 daemon.err uhttpd[1277]: cat: can't open '/etc/bench.log': No such file or directory
Wed Nov 30 20:32:05 2022 kern.info kernel: [27124.967273] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 20:32:05 2022 kern.info kernel: [27124.968014] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 20:32:05 2022 kern.info kernel: [27124.968609] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 20:32:05 2022 kern.info kernel: [27124.969225] r8125 0001:01:00.0 eth0: rss get rxnfc
Wed Nov 30 20:32:05 2022 kern.info kernel: [27124.970466] r8125 0001:01:00.0 eth0: rss get rxnfc

closed 2022-12-03 by AutoCONFIG

miaoermua, 2022-12-10:
老哥怎么解决的，是插件冲突吗

```
Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.780005] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.780879] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.781515] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.782335] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.782969] r8125 0002:01:00.0 eth1: rss get rxnfc
```

AutoCONFIG, 2022-12-15:
> 老哥怎么解决的，是插件冲突吗
>
> ```
> Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.780005] r8125 0002:01:00.0 eth1: rss get rxnfc
> Sat Dec 10 12:24:41 2022 kern.info kernel: [  245.780879] r8125 0002:01:00.0 [...]

您上面提供的这个日志是正常的网卡驱动信息，并不是错误报告。

我的这种问题，是因为我是用在校园网环境中（锐捷802.1x+PPPoE）,长时间大流量下载会掉线（多数是因为锐捷方面，有时也会pppoe掉线），主要问题在于认证插件方面没有保活检测机制，就是说因为网线松动或者网络质量堪忧掉线了，锐捷认证插件也无法检测到，几乎排除了硬件或lede的问题。

这种问题并非全是认证插件的问题，而是锐捷保活机制是单向的，认证客户端定时发送keep alive保活包给服务器，而服务器却不会发送任何反馈包，所以不能通过检测服务器的保活包来判断认证是否还在线。
其次，此数据包是基于UDP通信的，数据包发送请求传递给kernel后，哪怕网线被拔出，kernel也只是将数据包存在缓冲区，不断尝试发送，而不会返回任何错误信息给认证客户端。而实时监测网络连接状态会比较占用硬件资源，linux中断开发我目前还在处在学习状态，于是暂时放弃了对现有认证客户端插件（minieap&&mentohust for openwrt）的bug修复。

不过我找到了一个更好的解决方案，如果您也是跟我一样用于校园网环境的话，可能会对您有帮助，以下是解决方法：
1、原理：openwrt的pppoe是掉线自动重连的，基于此，可以利用pppoe掉线自动重连，顺带自动重连锐捷认证。因为pppoe或锐捷认证任意一个掉线，最终结果都是pppoe掉线。
2、为此，可以找到pppoe连接用的ppp脚本：/lib/netifd/proto/ppp.sh，使用vim或其他编辑器打开它
3、大概在219行处，有一个proto_pppoe_setup()函数，此函数里面有一个“/sbin/modprobe -qa slhc ppp_generic pppox pppoe ”，在其下方插入一段minieap/mentohust的认证命令"minieap -u用户名 -p密码……“
4、在254行附件有proto_pppoe_teardown()函数，在函数内插入一段”killall minieap“
5、以上修改完毕，保存退出即可
6、在管理页后台，网络-接口处删除wan6接口，因为此接口有时会干扰pppoe掉线自动重连功能
综上，所有操作完毕，重启路由器即可。

以上是我的解决方案，到目前为止再也没有出现过掉线的问题。希望对您有所帮助

miaoermua, 2022-12-15:
az 这太奇怪了，一样的网卡电犀牛会反馈这个，友善 r5s 不会反馈这个，感谢我后面测试一下会不会存在掉线问题👍

AutoCONFIG, 2022-12-15:
> az 这太奇怪了，一样的网卡电犀牛会反馈这个，友善 r5s 不会反馈这个，感谢我后面测试一下会不会存在掉线问题👍

好，也不完全排除是硬件设计或软件的问题，毕竟我之前用的x86软路由，在同样的网络环境下长时间大流量下载，从未出现过掉线的问题，也就是近期换成r66s之才出现的。
本来是想找店家保修的，排除硬件方面的问题，但是奈何店家已经跑路，无处可保。
不过好在我目前已经找到了好的解决方案，虽然不治本，但是掉线后能立即重连，能做到全程无感，也就这样心满意足了。

miaoermua, 2022-12-20:
> > rss get rxnfc
> 
> 这个信息应该是从网卡驱动发出来的，可以看一下网卡驱动源码，搜索这串字符串，看一下上下行代码，就比较了然了。
> 
> 最近没有什么空余时间，也就不在乎这些kernel.info信息，至少从没影响过正常使用。总言之能稳定使用就行。

好的，感谢正在作为主路由测试应该没问题

luyvlei, 2023-02-22:
我仅使用pppoe，也出现了大流量下载时断网的问题，内网能够访问路由器说明机器没有宕机，但是无法上外网，wan口并没有自动重播，删除wan6对断网自动重播有帮助吗

AutoCONFIG, 2023-02-23:
> 我仅使用pppoe，也出现了大流量下载时断网的问题，内网能够访问路由器说明机器没有宕机，但是无法上外网，wan口并没有自动重播，删除wan6对断网自动重播有帮助吗

建议尝试使用in-kernel的r8169驱动代替lede自带的r8125第三方驱动，因为我在使用r8169驱动时，一直都很稳定，而r8125会出现某些问题。当然不要看r8169驱动名称而以为是r8169专用驱动，实际上这是包括但不限于r8125/r8168/r8169等一系列网卡的专用驱动，是完美支持r8125网卡的（具体详情请在kernel查找）。

删除wan6能不能对自动重播有帮助，我目前不太确定，欢迎尝试，因为保留wan6，网卡也会自动重播，可能跟网络环境有关。

luyvlei, 2023-02-23:
> 建议尝试使用in-kernel的r8169驱动代替lede自带的r8125第三方驱动，因为我在使用r8169驱动时，一直都很稳定，而r8125会出现某些问题。当然不要看r8169驱动名称而以为是r8169专用驱动，实际上这是包括但不限于r8125/r8168/r8169等一系列网卡的专用驱动，是完美支持r8125网卡的（具体详情请在kernel查找）。 [...]

谢谢分享。我最近先排查一下是否是电信的问题，因为我推测电信可能是根据累计流量来触发断网重新拨号的，因为发生大流量断网的时候都是路由器开机较长时间未重启且当时正在下载中，电信对端可能把我的ip释放掉了。但是默认状态下openwrt的lcp是关闭的，我打开了lcp正在跑下载测试是否会触发重播