coolsnowwolf/lede #10534: 怎么最近几天编译的版本dhcp出问题了？

smdx, 2022-12-01:
各位好，在11月20来号开始编译的好几个版本dhcp都不能正常分配ip。疫情期间网络使用需求紧张就没细究直接在pve恢复旧备份了。今天又更新了昨晚编译的版本还是不能正常分配ip，云编译的配置没变动，op的设置都是直接恢复备份的。请问有大佬知道什么原因吗？

starxu1971, 2022-12-01:
大概率 广告屏蔽大师惹的祸，只要一开启就不能DHCP，关了就好。或者直接去掉这个软件。

smdx, 2022-12-01:
我没编译广告屏蔽大师这个插件，貌似是大佬更新的Dnsmasq造成的刚才对比了备份配置和固件包的配置格式不一样。
等网络空闲了再刷一次重新设置试试

starxu1971, 2022-12-01:
2.87的Dnsmasq也有问题，现在回退到2.86了。

smdx, 2022-12-02:
刚刚又重新刷了一遍固件，什么都没设置试连wifi还是分配不了ip......

hgh188, 2022-12-04:
一样的问题

Ginua8292, 2022-12-04:
问题依旧！只要开了SmartDNS和passwall,别说重启Dnsmasq，重启机器都不行

smdx, 2022-12-04:
我新刷的固件没开SmartDNS和passwall都是分配不了ip

rongcenter, 2022-12-05:
我有个不成熟的想法，会不会80内核有问题。

smdx, 2022-12-06:
昨晚编译了新版本今天刷了还是不行呢。。。。。。在1楼放上op日志了。有大佬知道原因吗？
[image]

closed 2022-12-20 by smdx

rsd1987, 2023-01-06:
是 procd jail的问题，屏蔽掉procd中的 dnsmasq相关jail的就可以了 。

smdx, 2023-01-06:
> 是 procd jail的问题，屏蔽掉procd中的 dnsmasq相关jail的就可以了 。

不用这么麻烦，找到方法在lxc conf
把常用的
lxc.cgroup2.devices.allow: c 108:0 rwm
lxc.cgroup2.devices.allow: c 10:200 rwm
改成
lxc.cgroup2.devices.allow: a
lxc.cap.drop:
一切都正常了，在lxc下openwrt也可以查看内核日志了。

hehe9011, 2023-03-23:
> 

亲测有效，感谢！！！

haohaoget, 2023-05-10:
太感谢了我搞了我几天原来是lxc配置的问题，一直以为是我openwrt的固件编译出问题了

flyhigao, 2024-05-29:
我的lxc openwrt也遇到dnsmasq无法启动问题,一开始我的解决方案是在rc.local加上:
`sleep 10 && /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid & ` 
后来看了openwrt.yaml发现可以这么解决:
`sed -i 's/procd_add_jail/# \0/g' /etc/init.d/dnsmasq`