coolsnowwolf/lede #10535: 网络接口页面数字溢出负数

deplives, 2022-12-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
网络接口页面的 数据包数据太大导致溢出

[image]

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

我自己扒了下代码，貌似是因为
https://github.com/coolsnowwolf/luci/blob/master/modules/luci-mod-admin-full/luasrc/view/admin_network/iface_status.htm#L34
中使用 %d 格式化 
https://github.com/coolsnowwolf/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/cbi.js#L1412
导致

不知道修改成 %s 能否解决这个问题，我试了一下，但是不知道为啥貌似没有生效。尝试删除luci的缓存无果

WYC-2020, 2022-12-01:
改那个不能防止rx_packets 这个溢出

deplives, 2022-12-01:
> 改那个不能防止rx_packets 这个溢出

大佬求教 这里应该怎么改。这里溢出看着真的是太难受了

WYC-2020, 2022-12-05:
https://github.com/openwrt/luci/issues/3003  这里参考改下cbi.js，然后把那个%d改成%s,我看了netifd里面用的无符号，所以哪里就不用管了最多会回环，不会出现负值，但是有可能后续会变成0.，这个没仔细研究过，你先改上面两个在观察看下吧

deplives, 2022-12-05:
> [openwrt/luci#3003](https://github.com/openwrt/luci/issues/3003) 这里参考改下cbi.js，然后把那个%d改成%s,我看了netifd里面用的无符号，所以哪里就不用管了最多会回环，不会出现负值，但是有可能后续会变成0.，这个没仔细研究过，你先改上面两个在观察看下吧

十分感谢提供的信息，我去研究研究

shoucandanghehe, 2022-12-05:
感觉需要达到一定数量之后换单位 不然还是会溢出（

closed 2026-09-05 by coolsnowwolf