coolsnowwolf/lede #10540: upx更新3.96后，在线编译出现错误

jliveusa, 2022-12-03:
2022-12-03T06:14:31.7764457Z make[3]: Entering directory '/workdir/openwrt/package/lean/upx'
2022-12-03T06:14:31.9557594Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/.built
2022-12-03T06:14:31.9583781Z touch /workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/.built_check
2022-12-03T06:14:31.9609859Z CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96=upx-3.96 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96=upx-3.96 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96=upx-3.96 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
2022-12-03T06:14:31.9644676Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/src'
2022-12-03T06:14:31.9697484Z x86_64-openwrt-linux-musl-g++ -I./../src/lzma-sdk -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96=upx-3.96 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -O2 -fno-delete-null-pointer-checks -fno-strict-aliasing -fwrapv -funsigned-char -Wall -W -Wcast-align -Wcast-qual -Wmissing-declarations -Wpointer-arith -Wshadow -Wvla -Wwrite-strings -Werror -Wno-shadow -o compress_lzma.o -c compress_lzma.cpp
2022-12-03T06:14:32.0955349Z In file included from ./../src/lzma-sdk/C/7zip/Compress/LZMA/LZMAEncoder.h:6,
2022-12-03T06:14:32.0955729Z                  from compress_lzma.cpp:235:
2022-12-03T06:14:32.0956910Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/LZMAEncoder.h: In member function 'virtual ULONG NCompress::NLZMA::CEncoder::Release()':
2022-12-03T06:14:32.0957536Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:159:32: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
2022-12-03T06:14:32.0958026Z   159 | STDMETHOD_(ULONG, Release)() { if (--__m_RefCount != 0)  \
2022-12-03T06:14:32.0958318Z       |                                ^~
2022-12-03T06:14:32.0958786Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:166:3: note: in expansion of macro 'MY_ADDREF_RELEASE'
2022-12-03T06:14:32.0959115Z   166 |   MY_ADDREF_RELEASE
2022-12-03T06:14:32.0959330Z       |   ^~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0959755Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:182:37: note: in expansion of macro 'MY_UNKNOWN_IMP_SPEC'
2022-12-03T06:14:32.0960124Z   182 | #define MY_UNKNOWN_IMP3(i1, i2, i3) MY_UNKNOWN_IMP_SPEC( \
2022-12-03T06:14:32.0960403Z       |                                     ^~~~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0960832Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/LZMAEncoder.h:371:3: note: in expansion of macro 'MY_UNKNOWN_IMP3'
2022-12-03T06:14:32.0961145Z   371 |   MY_UNKNOWN_IMP3(
2022-12-03T06:14:32.0961349Z       |   ^~~~~~~~~~~~~~~
2022-12-03T06:14:32.0961850Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:160:24: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
2022-12-03T06:14:32.0962252Z   160 |   return __m_RefCount; delete this; return 0; }
2022-12-03T06:14:32.0962725Z       |                        ^~~~~~
2022-12-03T06:14:32.0963162Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:166:3: note: in expansion of macro 'MY_ADDREF_RELEASE'
2022-12-03T06:14:32.0963478Z   166 |   MY_ADDREF_RELEASE
2022-12-03T06:14:32.0963682Z       |   ^~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0964095Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:182:37: note: in expansion of macro 'MY_UNKNOWN_IMP_SPEC'
2022-12-03T06:14:32.0964463Z   182 | #define MY_UNKNOWN_IMP3(i1, i2, i3) MY_UNKNOWN_IMP_SPEC( \
2022-12-03T06:14:32.0964749Z       |                                     ^~~~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0965177Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/LZMAEncoder.h:371:3: note: in expansion of macro 'MY_UNKNOWN_IMP3'
2022-12-03T06:14:32.0965583Z   371 |   MY_UNKNOWN_IMP3(
2022-12-03T06:14:32.0965787Z       |   ^~~~~~~~~~~~~~~
2022-12-03T06:14:32.0972423Z compress_lzma.cpp: In member function 'virtual ULONG MyLzma::InStream::Release()':
2022-12-03T06:14:32.0973241Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:159:32: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
2022-12-03T06:14:32.0974210Z   159 | STDMETHOD_(ULONG, Release)() { if (--__m_RefCount != 0)  \
2022-12-03T06:14:32.0974517Z       |                                ^~
2022-12-03T06:14:32.0975003Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:171:3: note: in expansion of macro 'MY_ADDREF_RELEASE'
2022-12-03T06:14:32.0975344Z   171 |   MY_ADDREF_RELEASE
2022-12-03T06:14:32.0975546Z       |   ^~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0975898Z compress_lzma.cpp:242:5: note: in expansion of macro 'MY_UNKNOWN_IMP'
2022-12-03T06:14:32.0976180Z   242 |     MY_UNKNOWN_IMP
2022-12-03T06:14:32.0976380Z       |     ^~~~~~~~~~~~~~
2022-12-03T06:14:32.0976908Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:160:24: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
2022-12-03T06:14:32.0977315Z   160 |   return __m_RefCount; delete this; return 0; }
2022-12-03T06:14:32.0977566Z       |                        ^~~~~~
2022-12-03T06:14:32.0977989Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:171:3: note: in expansion of macro 'MY_ADDREF_RELEASE'
2022-12-03T06:14:32.0978302Z   171 |   MY_ADDREF_RELEASE
2022-12-03T06:14:32.0978502Z       |   ^~~~~~~~~~~~~~~~~
2022-12-03T06:14:32.0978834Z compress_lzma.cpp:242:5: note: in expansion of macro 'MY_UNKNOWN_IMP'
2022-12-03T06:14:32.0979094Z   242 |     MY_UNKNOWN_IMP
2022-12-03T06:14:32.0979293Z       |     ^~~~~~~~~~~~~~
2022-12-03T06:14:32.0989405Z compress_lzma.cpp: In member function 'virtual ULONG MyLzma::OutStream::Release()':
2022-12-03T06:14:32.0990031Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:159:32: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
2022-12-03T06:14:32.0990527Z   159 | STDMETHOD_(ULONG, Release)() { if (--__m_RefCount != 0)  \
2022-12-03T06:14:32.0990803Z       |                                ^~
2022-12-03T06:14:32.0991223Z ./../src/lzma-sdk/C/7zip/Compress/LZMA/../../../Common/MyCom.h:171:3: note: in expansion of macro 'MY_ADDREF_RELEASE'
[... cut ...]
2022-12-03T06:14:34.0588332Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/src'
2022-12-03T06:14:34.0589031Z make[3]: Leaving directory '/workdir/openwrt/package/lean/upx'
2022-12-03T06:14:34.0591611Z make[3]: *** [Makefile:65: /workdir/openwrt/build_dir/target-x86_64_musl/upx-3.96/.built] Error 2
2022-12-03T06:14:34.0594893Z time: package/lean/upx/compile#2.20#0.09#2.29
2022-12-03T06:14:34.0602721Z     ERROR: package/lean/upx failed to build.
2022-12-03T06:14:34.0612132Z make[2]: *** [package/Makefile:116: package/lean/upx/compile] Error 1

zgc, 2022-12-03:
真机编译也失败

closed 2022-12-03 by aiamadeus

sinaamos, 2023-04-27:
> 真机编译也失败

解決了嗎？同樣問題  upx-3.96.tar.xz: Download from https://github.com/upx/upx.git failed

commit 276348a4 (2022-12-03): upx: fixes build with gcc 11

Fixes: #10540
Fixes: a5c9813 ("upx: witch to the release version")