coolsnowwolf/lede #10545: 拉取最新源码后编译报错

bi4tms, 2022-12-04:
拉取最新源码，编译x86固件报错，代码如下：
make[6]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/src'
make[5]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/src'
Making install in examples
make[5]: Entering directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/examples'
make[6]: Entering directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/examples'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/examples'
make[5]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/examples'
make[5]: Entering directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12'
make[6]: Entering directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12'
make[5]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12'
make[4]: Leaving directory '/home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12'
touch /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.built
rm -rf /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3.installed /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3
mkdir -p /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3
install -d -m0755 /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3/usr/bin
install -m0755 /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/ipkg-install/usr/bin/iperf3 /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3/usr/bin/
touch /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3.installed
mkdir -p /home/x86.6.0/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/x86.6.0/lede/tmp/.root-copy.flock -c 'cp -fpR /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3/. /home/x86.6.0/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/x86.6.0/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.iperf3_installed
removed '/home/x86.6.0/lede/bin/packages/x86_64/packages/libiperf3_3.12-1_x86_64.ipk'
mkdir -p /home/x86.6.0/lede/bin/targets/x86/64/packages /home/x86.6.0/lede/build_dir/target-x86_64_musl/iperf-nossl/iperf-3.12/ipkg-x86_64/libiperf3/CONTROL /home/x86.6.0/lede/staging_dir/target-x86_64_musl/pkginfo
ifeq (nossl,ssl)
bash: -c: line 0: syntax error near unexpected token `nossl,ssl'
bash: -c: line 0: `ifeq (nossl,ssl)'
make[3]: *** [Makefile:113: /home/x86.6.0/lede/bin/packages/x86_64/packages/libiperf3_3.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/x86.6.0/lede/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/nossl/compile#19.43#19.78#41.64
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: nossl).
make[2]: *** [package/Makefile:116: package/feeds/packages/iperf3/compile] Error 1
make[2]: Leaving directory '/home/x86.6.0/lede'
make[1]: *** [package/Makefile:110: /home/x86.6.0/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.6.0/lede'
make: *** [/home/x86.6.0/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2022-12-04:
`
ERROR: package/feeds/packages/iperf3 failed to build (build variant: nossl).
`
换个其他版本的iperf3编译下？

firkerword, 2022-12-05:
同样的错误

yanxin152133, 2022-12-05:
不要那个插件不就行了，编译完手动安装

wangweitung, 2022-12-05:
> […](#)

需要取消哪个插件？

yanxin152133, 2022-12-05:
iperf3

jdnt, 2022-12-05:
> […](#)

一时没想起来这就是个测速的，装不装根本无所谓……感谢提醒！

WYC-2020, 2022-12-05:
看另外一个帖子

closed 2022-12-05 by coolsnowwolf