coolsnowwolf/lede #10547: 刚git的包，feeds更新时报警告，提示v2ray-geoip/geosite不存在

siestina, 2022-12-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：feeds更新报错，怕有什么隐患存在

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：./scripts/feeds install -a
WARNING: Makefile 'package/feeds/luci/luci-app-mosdns/Makefile' has a dependency on 'v2ray-geoip', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-mosdns/Makefile' has a dependency on 'v2ray-geosite', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

lovehackintosh, 2022-12-05:
```bash
sed -i 's/#src-git helloworld/src-git helloworld/g' ./feeds.conf.default
./scripts/feeds update -a && ./scripts/feeds install -a
```

closed 2022-12-05 by aiamadeus

shaojs321, 2022-12-17:
> ```shell
> sed -i 's/#src-git helloworld/src-git helloworld/g' ./feeds.conf.default
> ./scripts/feeds update -a && ./scripts/feeds install -a
> ```

好用