coolsnowwolf/lede #10549: 最新源码，iperf3无法通过编译，内有日志错误。

a904055262, 2022-12-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
pkgdir/gzip/. /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.gzip_installed
echo "gzip" >> /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/gzip.default.install
make[3]: Leaving directory '/home/nlf/lede/feeds/packages/utils/gzip'
time: package/feeds/packages/gzip/compile#0.08#0.04#0.11
make[3]: Entering directory '/home/nlf/lede/package/libs/sysfsutils'
make[3]: Leaving directory '/home/nlf/lede/package/libs/sysfsutils'
time: package/libs/sysfsutils/compile#0.09#0.04#0.12
make[3]: Entering directory '/home/nlf/lede/feeds/packages/utils/lm-sensors'
make[3]: Leaving directory '/home/nlf/lede/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.08#0.02#0.10
make[3]: Entering directory '/home/nlf/lede/feeds/packages/admin/htop'
mkdir -p /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp
SHELL= flock /home/nlf/lede/tmp/.root-copy.flock -c 'cp -fpR /home/nlf/lede/build_dir/target-aarch64_cortex-a53_musl/htop-3.2.1/.pkgdir/htop/. /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.htop_installed
echo "htop" >> /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/htop.default.install
make[3]: Leaving directory '/home/nlf/lede/feeds/packages/admin/htop'
time: package/feeds/packages/htop/compile#0.09#0.03#0.11
make[3]: Entering directory '/home/nlf/lede/feeds/packages/net/iperf3'
mkdir -p /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp
SHELL= flock /home/nlf/lede/tmp/.root-copy.flock -c 'cp -fpR /home/nlf/lede/build_dir/target-aarch64_cortex-a53_musl/iperf-nossl/iperf-3.12/.pkgdir/iperf3/. /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.iperf3_installed
mkdir -p /home/nlf/lede/bin/targets/armvirt/64/packages /home/nlf/lede/build_dir/target-aarch64_cortex-a53_musl/iperf-nossl/iperf-3.12/ipkg-aarch64_cortex-a53/libiperf3/CONTROL /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo
ifeq (nossl,ssl)
bash: -c: line 0: syntax error near unexpected token `nossl,ssl'
bash: -c: line 0: `ifeq (nossl,ssl)'
make[3]: *** [Makefile:113: /home/nlf/lede/bin/packages/aarch64_cortex-a53/packages/libiperf3_3.12-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/nlf/lede/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/nossl/compile#0.09#0.04#0.12
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: nossl).
make[2]: *** [package/Makefile:116: package/feeds/packages/iperf3/compile] Error 1
make[2]: Leaving directory '/home/nlf/lede'
make[1]: *** [package/Makefile:110: /home/nlf/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nlf/lede'
make: *** [/home/nlf/lede/include/toplevel.mk:231：world] 错误 2

A：

去掉iperf3能编译成功。gitaction在线编译也是卡在iperf3过不去

WYC-2020, 2022-12-05:
[image]
这个提交导致，我是想不通有ifneq 为啥要去用ifeq然后留一个空白的if分支，改用ifneq 就行了

a904055262, 2022-12-05:
这位大佬提交代码都不带测试的。。

WYC-2020, 2022-12-05:
改成这样
[image]

亲测

wangweitung, 2022-12-05:
https://github.com/coolsnowwolf/packages/commit/d454ebbdc859ef2f57e5aa605436d01059e1766e#diff-71f4db07398d8c351a50cd0bde79fd9a78ba95d80b0d908e2657b63e8f0088f2

ximiTech, 2022-12-05:
为什么我直接点commits看不到这次提交呢？要怎么看的？

closed 2022-12-05 by coolsnowwolf

africa1207, 2022-12-06:
> 改成这样 [image]
> 
> 亲测

全局也没搜索到iperf3文件夹，能否给个路径

zzy19868, 2022-12-07:
> > 改成这样 [image]
> > 亲测
> 
> 全局也没搜索到iperf3文件夹，能否给个路径

已经可以正常编译了，拉取最新源码。