coolsnowwolf/lede #10555: aarch64固件的python3编译不通过

HoldOnBro, 2022-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：aarch64固件的python3编译不通过

### (2) 路由器型号和固件版本

A：aarch64_cortex-a53，最新源码

### (3) 详细日志

A：
```
if [ -d "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/bin" ]; then 	/workdir/openwrt/staging_dir/host/bin/sed -i -e "1"'!'"b;s,^#"'!'".*python.*,#"'!'"/usr/bin/python3," -i --follow-symlinks /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/bin/* ; fi
/workdir/openwrt/staging_dir/host/bin/sed: cannot rename /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/bin/sedwCWeic: Not a directory
make[3]: *** [Makefile:405: /workdir/openwrt/bin/packages/aarch64_cortex-a53/packages/python3-base_3.10.7-1_aarch64_cortex-a53.ipk] Error 4
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#0.66#0.16#0.77
    ERROR: package/feeds/packages/python3 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

kissyouhunter, 2022-12-08:
还没有修复呢吧
昨天试了编译一次
还是同样的报错

HoldOnBro, 2022-12-08:
> 还没有修复呢吧 昨天试了编译一次 还是同样的报错

应该还没修

coolsnowwolf, 2022-12-08:
你们不是旁观者，如果找到的修复的方法发个 PR

rq1025330, 2022-12-10:
> 
大佬你是sed 回滚的那个文件呀

Jejz168, 2022-12-12:
> > 
> 
> 大佬你是sed 回滚的那个文件呀
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=4.8/g' tools/sed/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile

rq1025330, 2022-12-13:
> > > 
> > 
> > 
> > 大佬你是sed 回滚的那个文件呀
> > sed -i 's/PKG_VERSION:=._/PKG_VERSION:=4.8/g' tools/sed/Makefile
> > sed -i 's/PKG_HASH:=._/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile

sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=4.8/g' tools/sed/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile
这个

rq1025330, 2022-12-13:
> > > 
> > 
> > 
> > 大佬你是sed 回滚的那个文件呀
> > sed -i 's/PKG_VERSION:=._/PKG_VERSION:=4.8/g' tools/sed/Makefile
> > sed -i 's/PKG_HASH:=._/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile

* 没显示出来

Jejz168, 2022-12-13:
> > > > 
> > > 
> > > 
> > > 大佬你是sed 回滚的那个文件呀
> > > sed -i 's/PKG_VERSION:=._/PKG_VERSION:=4.8/g' tools/sed/Makefile
> > > sed -i 's/PKG_HASH:=._/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile
> > > 星号 ，没显示出来
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=4.8/g' tools/sed/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile

Jejz168, 2022-12-13:
> > > > 
> > > 
> > > 
> > > 大佬你是sed 回滚的那个文件呀
> > > sed -i 's/PKG_VERSION:=._/PKG_VERSION:=4.8/g' tools/sed/Makefile
> > > sed -i 's/PKG_HASH:=._/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile
> > > 星号 ，没显示出来

还真是，大意了

HoldOnBro, 2022-12-15:
> > > 
> > 
> > 
> > 大佬你是sed 回滚的那个文件呀
> > sed -i 's/PKG_VERSION:=._/PKG_VERSION:=4.8/g' tools/sed/Makefile
> > sed -i 's/PKG_HASH:=._/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g' tools/sed/Makefile
> > 星号 ，没显示出来
> > PKG_VERSION:=.星号
> > PKG_HASH:=.星号
```
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=4.8/g' tools/sed/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=f79b0cfea71b37a8eeec8490db6c5f7ae7719c35587f21edb0617f370eeff633/g'
```

好使

lovehackintosh, 2022-12-19:
解决方法: https://github.com/openwrt/openwrt/commit/c2000ebfa53f2234f662f85109890e80e39b70ba

commit bcfa26c0 (2022-09-12): tools: remove xxd package

It shouldn't be needed anymore as we've now `scripts/xxdi.pl`, which
should be self contained and fully compatible `xxd -i` replacement.

Fixes: #10555
commit fa4dc514 (2022-09-12): scripts: xxdi.pl: add xxd -i compat mode

So it can serve as a standalone drop in replacement for xxd utility used
currently mostly in U-Boot packages with `xxd -i` mode which outputs C
include file style, with aim for byte to byte identical output, so the
eventual difference in the generated output is easily spottable.

Fixes: #10555
PR #10642 (2022-12-19): tools/sed: Fix handling of symlinks of 128 chars
If the absolute path a symlink is pointing to is 128 bytes long sed failed with an error message like this: "<path>/sedstbU8O: Not a directory"

This fixes a problem building python seen in the build bot.

This patch is on its way into upstream sed.

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
Fixes: #10555, #10580
closed 2022-12-19 by coolsnowwolf