coolsnowwolf/lede #10557: 云编译固件时出错，Package lib32gcc1 is not available

oldskooler, 2022-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
.
.
.
Get:27 http://azure.archive.ubuntu.com/ubuntu jammy-security/restricted Translation-en [68.7 kB]
Get:28 http://azure.archive.ubuntu.com/ubuntu jammy-security/restricted amd64 c-n-f Metadata [532 B]
Get:29 http://azure.archive.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [622 kB]
Get:30 http://azure.archive.ubuntu.com/ubuntu jammy-security/universe Translation-en [81.5 kB]
Get:31 http://azure.archive.ubuntu.com/ubuntu jammy-security/universe amd64 c-n-f Metadata [11.0 kB]
Get:32 http://azure.archive.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [4268 B]
Get:33 http://azure.archive.ubuntu.com/ubuntu jammy-security/multiverse Translation-en [972 B]
Get:34 http://azure.archive.ubuntu.com/ubuntu jammy-security/multiverse amd64 c-n-f Metadata [228 B]
Fetched 4631 kB in 1s (4673 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Package lib32gcc1 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  lib32gcc-s1

E: Package 'lib32gcc1' has no installation candidate
Error: Process completed with exit code 100.

q405812306, 2022-12-06:
要么用ubuntu18.04，要么用ubuntu-latest之下把update的依赖换成L大md文件中写的，也就是
sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev）

HoangNghi, 2022-12-07:
> 要么用ubuntu18.04，要么用ubuntu-latest之下把update的依赖换成L大md文件中写的，也就是 sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential bzip2 ccache cmake cpio curl [...]

换成L大md文件中写的依赖之后反而提示是ack出错了

q405812306, 2022-12-07:
> 换成L大md文件中写的依赖之后反而提示是ack出错了

你看我仓库fork的lede的yml

oldskooler, 2022-12-07:
我把openwrt-ci.yml文件里面的下面这一行：
sudo -E apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler antlr3 gperf swig
改成这样后，云编译可以完成了
sudo -E apt-get -y install ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential \
                           bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
                           git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
                           libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
                           mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils \
                           rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev