coolsnowwolf/lede #10558: 网页服务是nginx时，升级固件上次失败！

sjkhsl, 2022-12-06:
上次固件失败，提示 
`502 Bad Gateway
 nginx/1.21.3 `

上传.img文件就会报错，只有上传.gz的压缩文件才能升级，可是上传.gz的文件并不能保存配置升级。

在nginx配置文件中修改这两项没有作用！
```
	client_max_body_size 2560M;
	large_client_header_buffers 200 100k;
```

WYC-2020, 2022-12-06:
上传gz也可以保存配置的 我每次都是这样搞的，至于那个nginx 我以前玩过，是可以的现在不知道什么情况了

sjkhsl, 2022-12-06:
> 上传gz也可以保存配置的 我每次都是这样搞的，至于那个nginx 我以前玩过，是可以的现在不知道什么情况了

我使用.gz，保留配置，然后重启后就成原始状态了～好奇怪～

ximiTech, 2022-12-07:
昨天开始的，我的情况和楼主一样，使用gz无法保存配置升级。。。

ximiTech, 2022-12-08:
我查出原因了，不能安装普通的gzip包，升级程序必须要使用busybox提供的gzip来解压。。。

sjkhsl, 2022-12-08:
> 我查出原因了，不能安装普通的gzip包，升级程序必须要使用busybox提供的gzip来解压。。。

怎么搞？busybox不是系统默认带的？

hyy-666, 2022-12-09:
貌似busybox外的gzip的锅，不要选就是，我也遇到过这个问题，最后看ttl查出来的，解压固件都没成功直接重启了

coolsnowwolf, 2026-09-05:
Nginx 502 上传问题属于反向代理请求体大小/超时与 LuCI 上传流程配置，报告没有完整 nginx 配置、错误日志或当前 LuCI 版本复现，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf