coolsnowwolf/lede #10559: 源码自带的transmission版本太旧，且不能打开web ui页面，换成新版后，页面中的启用功能失效了

xaisc, 2022-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
Transmission 部分编译时，报告版本和hash不对，貌似已经升级了

transmission-web-control-2020-09-13-c26a0761.tar.xz: Download from https://github.com/ronggang/transmission-web-control failed
transmission-web-control-2020-09-13-c26a0761.tar.xz: Wrong hash (probably caused by .gitattributes), expecting c078750537e16cd2dd1f6d01f81d1cba5b2e3a9736839970c607402473adfca0, got 267a7bf06c093668c2f9d5c179ea45bee624cee250d8bdcb0073dfe98b4af679
Checking out files from the git repository...
Cloning into 'transmission-web-control-2020-09-13-c26a0761'...
remote: Enumerating objects: 4700, done.
remote: Counting objects: 100% (155/155), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 4700 (delta 96), reused 121 (delta 74), pack-reused 4545
Receiving objects: 100% (4700/4700), 21.15 MiB | 754.00 KiB/s, done.
Resolving deltas: 100% (3184/3184), done.
Note: switching to 'c26a0761e3a8fe3cff2480735ec363dc253c5105'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at c26a076 releases: v1.6.1
Packing checkout...

### (1) 具体问题

A：源码中自带的transmission版本太低了，编译时报错，编译完启动transmission后，无法打开管理web页面：

404: Not Found
Couldn't find Transmission's web interface files!

Users: to tell Transmission where to look, set the TRANSMISSION_WEB_HOME environment variable to the folder where the web interface's index.html is located.

Package Builders: to set a custom default at compile time, #define PACKAGE_DATA_DIR in libtransmission/platform.c or tweak tr_getClutchDir () by hand.

重新编译，手动改成transmission 3.0以后，luci-app-transmission不能启动transimission服务了（页面上的那个启用功能失效了），需要手动通过transmission start。手动启动后，web ui的页面能正常过访问
后面这个启用功能失效的错误好像有人提到过
### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

WYC-2020, 2022-12-08:
这玩意我记得以前我用过 没问题，现在有问题了？但是我看最近那玩意也没怎么更新，我怀疑是更新了什么另外的库导致的

WYC-2020, 2022-12-08:
[image]
 这样修复亲测有效

WXfeng-boy, 2023-01-02:
你好，请问你这个问题解决了吗

coolsnowwolf, 2026-09-05:
已参照 OpenWrt 23.05 更新并 push：packages `e07864161072` 更新 transmission-web-control 到 2021 snapshot 并安装到 `public_html`；`50d3d789451e` 更新 transmission 默认 `web_home` 路径。这样修复旧 web-control hash/目录不匹配导致的 404 与启用失败，关闭为 completed。

closed 2026-09-05 by coolsnowwolf